openhardwaremonitor/openhardwaremonitor #988: MSI 970 Gaming (MS-7693) motherboard has no sensors shown

Seksys, 2017-06-10:
Hello, I've glanced at the source thinking perhaps I could just add support myself, but perhaps I was a bit too naive. The MSI supported utilities show sensors as does the BIOS. It's an AMD chipset, and basic info from MSI: it's using AMD 970 and SB950 chipset. If the support utility can grab the info I presume OHM can too.

And this is the link directly from the manufacturer:
[MSI website](https://us.msi.com/Motherboard/970-GAMING.html#hero-specification)

Here's my report:
[MSI970Gaming.MS-7693.OpenHardwareMonitor.Report.txt](https://github.com/openhardwaremonitor/openhardwaremonitor/files/1066108/MSI970Gaming.MS-7693.OpenHardwareMonitor.Report.txt)

I hope this helps get support in future versions.

DarthDestroyer, 2017-06-10:
http://openhardwaremonitor.org/files/openhardwaremonitor-v0.8.0.3-alpha.zip

Your Mainboard Support is not pending, at the moment!
(feel free to implement it by yourself)

Regards

For further investigations:
Chip ID: Unknown Winbond / Nuvoton / Fintek with ID 0x1106 at 0x4E/0x4F

mykolq, 2017-06-13:
Sorry, can you show how can i implement new hardware by myself?