openhardwaremonitor/openhardwaremonitor #990: CPU temperature values too high if writing to a report

knmurf, 2017-06-14:
The CPU temperatures given by OpenHardwareMonitor (individual core temperatures and package temperature) all about 31 degrees Celcius. This equals other tools and seems to be correct.

However if I use the following small program to just write values to a file then the CPU temperature values are significantly higher, for example 47 degrees. 

[image]
[temperature.txt](https://github.com/openhardwaremonitor/openhardwaremonitor/files/1074629/temperature.txt)

What is wrong?

using System.Globalization;
using System.IO;
using OpenHardwareMonitor.Hardware;

namespace OpenHardwareMonitorReport
{
  class Program
  {
    static void Main(string[] args)
    {
      Computer computer = new Computer();

      computer.Open();

      computer.MainboardEnabled = true;
      computer.CPUEnabled = true;
      computer.RAMEnabled = true;
      computer.GPUEnabled = true;
      computer.HDDEnabled = true;

      var report = computer.GetReport();
      File.AppendAllText(@"temperature.txt", string.Format(CultureInfo.InvariantCulture, report));

      computer.Close();
    }
  }
}

DarthDestroyer, 2017-06-14:
you will get peaks always! 

the problem of your way is: open computer, getreport and then close Computer. 
That means, you will get one sensorcheck. that is visible because min = max = value! 
The peak of 47 degree looks realistic for a thread using that core (i would say, that ohm query). 

try to get multiple results, that could lead to a Solution! 
put a Computer.Update() infront of the getreport... that creates two results... 
that should help at that way, maybe

knmurf, 2017-06-15:
Calling update() helps. I rewrote my program:

using System.Globalization;
using System.IO;
using System.Threading;
using OpenHardwareMonitor.Hardware;

namespace OpenHardwareMonitorReport
{
  class Program
  {
    static void Main(string[] args)
    {
      // ReSharper disable once ObjectCreationAsStatement
      new Run(args);
    }
  }

  internal class Run
  {
    readonly Computer computer = new Computer();

    public Run(string[] args)
    {
      computer.Open();

      computer.MainboardEnabled = true;
      computer.CPUEnabled = true;
      computer.RAMEnabled = true;
      computer.GPUEnabled = true;
      computer.HDDEnabled = true;

      var timer = new Timer(OnTimer);
      timer.Change(100, 100);

      Thread.Sleep(500);

      var report = computer.GetReport();
      var outputFile = args != null && args.Length > 0 && !string.IsNullOrEmpty(args[0])
        ? args[0]
        : @"OpenHardWareMonitorReport.txt";
      File.WriteAllText(outputFile, string.Format(CultureInfo.InvariantCulture, report));

      timer.Dispose();
      Thread.Sleep(30);
      computer.Close();
    }

    private void OnTimer(object state)
    {
      computer.Accept(new UpdateVisitor());
    }
  }

  public class UpdateVisitor : IVisitor
  {
    public void VisitComputer(IComputer computer)
    {
      computer.Traverse(this);
    }

    public void VisitHardware(IHardware hardware)
    {
      hardware.Update();

      foreach (IHardware subHardware in hardware.SubHardware)
      {
        subHardware.Accept(this);
      }
    }

    public void VisitSensor(ISensor sensor)
    {
    }

    public void VisitParameter(IParameter parameter)
    {
    }
  }
}

closed 2017-06-15 by knmurf