openhardwaremonitor/openhardwaremonitor #991: I Read a lot about 0.8.0.3 Alpha but Where I Download it ?

Ciberti, 2017-06-15:
Hello Thank in advance for reading this.

I Read a lot about the 0.8.0.3 Alpha Version of  the Open Hardware Monitor, My System is not Getting Any Info from the 0.8.0 Beta Version.

Could you please let me know from where download the 0.8.0.3 Alpha Version?

I'm Uploading as well a Report.

Thanks a lot,
Benny Gomez
[OpenHardwareMonitor.Report.txt](https://github.com/openhardwaremonitor/openhardwaremonitor/files/1078282/OpenHardwareMonitor.Report.txt)

DarthDestroyer, 2017-06-15:
Two Issues down = http://openhardwaremonitor.org/files/openhardwaremonitor-v0.8.0.3-alpha.zip

Hope that helps! Regards

Ciberti, 2017-06-15:
Thanks a lot :) !! that Worked Just Fine !! 👍