openhardwaremonitor/openhardwaremonitor #992: Dark UI/window background mode

JonnyRedHed, 2017-06-20:
Having this open on a spare monitor all the time at night can be harsh on the eyes, when next to the dark background of Afterbuner's graph view.  

Could I suggest a dark UI mode (default off) inc window background, with dull/medium bright white text.  Easy on the night time eyes.

jfarmersp, 2019-01-18:
Would love this.

JonnyRedHed, 2019-01-19:
I was thinking along the lines of Discord for windows, where the UI window frame is the same dark as the internal area.  Not just a dark skin inside a bright windows UI,  where the menus and X and min and max buttons.  So its a proper night time theme like Discord uses.

jfarmersp, 2019-01-22:
Definitely an important distinction regarding the menu bar, apps with dark modes/themes that miss that piece are really frustrating! Love the decisions Discord made regarding their UI, would love to see something similar become available with OHM.

JonnyRedHed, 2019-01-22:
The reason for me is simple, I use an older LCD 2nd monitor to the side of my large main TV,  this smaller LCD shows MSI afterburner (original dark skin) with stats window (dark),  and openhardwharemonitor.   The white of the current version distracts.   So I'm all for dark mode window and UI's for everything these days.

DarthDestroyer, 2019-01-22:
You got me already with that idea ;-) it would be great, for sure!

but you should try it out to change it by hand! after that, you will know what i mean ;-) 
and sadly, windows forms arent that free to design :-/

JonnyRedHed, 2019-03-19:
Any update on this request.  Would be real nice to have this option. Its been years.