openhardwaremonitor/openhardwaremonitor #993: ASUS CROSSHAIR VI HERO

mwebi, 2017-06-20:
Hey,

the ASUS CROSSHAIR VI HERO is only displayed as a headline in OpenHardwareMonitor. Would it be possible to read the sensors (there's several like water pump RPM, water pump flow in/out, temp sensor etc?)

Thanks!

DarthDestroyer, 2017-06-21:
Your Mainboard isnt supported, at the moment! 

It would help, if you could upload a full report!
Thanks and Regards

mwebi, 2017-06-21:
Sure thing:

[OpenHardwareMonitor.Report.txt](https://github.com/openhardwaremonitor/openhardwaremonitor/files/1092140/OpenHardwareMonitor.Report.txt)

BennehBoy, 2017-07-13:
There's a WIP branch here that adds preliminary support for the IT8665 -> https://github.com/azuisleet/openhardwaremonitor/commit/d4b763c01625d595e22b2b658e55696827b44dac#commitcomment-23040901

moellerm, 2020-03-08:
The ITE super I/O chip IT8665E used on this mainboard should be supported in Open Hardware Monitor Version 0.9.2.

closed 2020-03-08 by moellerm