openhardwaremonitor/openhardwaremonitor #995: Ryzen 5 support

jakobx13, 2017-06-21:
[OpenHardwareMonitor.Report.txt](https://github.com/openhardwaremonitor/openhardwaremonitor/files/1092691/OpenHardwareMonitor.Report.txt)

DarthDestroyer, 2017-06-21:
#976

jakobx13, 2017-07-06:
What is this

DarthDestroyer, 2017-07-06:
I think, it is that, what you had requested...

jakobx13, 2017-07-06:
@DarthDestroyer I'm sorry im a noob idk what im supposed to do i have not used the site before

DarthDestroyer, 2017-07-06:
that problem is solveable... but no one can help you, if you dont write what you want and what you could do! 
i would live to see atleast one sentence of text, if you have a issue in the future... ;) 

that, what i found in your report is solved / discussed in the link i had posted! 
if you dont know what you have to do, it will get complicate! 

you have to download the Sourcecode, make these changes (it is in the link) and then compile it! 

if you are not familar with that or not willing to learn that, your last and only hope is - waiting until it gets included in the next update :/

that sounds hard, but the possibilities are limited :/

jakobx13, 2017-07-06:
I am Much willing to learn i do have an interest in coding @DarthDestroyer

DarthDestroyer, 2017-07-06:
That is the first Step ;-)

* Now, get Visual Studio Community Edition and install it. 
* Then, Download OHM Sourcecode. 
* Open it with VS
* Look at the Explorer and learn to navigate
* click on the link i had posted
* switch to "changes" tab
* do anything what there is noted, in the files the, are listed
(red = delete, green = add) 
* compile it while pressing "Run" 
* #success

;)

jakobx13, 2017-07-06:
i'll give it a shot @DarthDestroyer

DarthDestroyer, 2017-07-06:
it isnt really hard and you will get the basics fast! 
i wish you fun and luck! ;)

jakobx13, 2017-07-06:
thank you @DarthDestroyer

jakobx13, 2017-07-06:
[image]
@DarthDestroyer witch one of these do i download

jakobx13, 2017-07-07:
okay umm i dont know where to find the .oem file i dowloaded the source code though

DarthDestroyer, 2017-07-07:
first one and i recommend the second, too! 

you have to open the project file in the root folder you had downloaded and installed.

jakobx13, 2017-07-07:
@DarthDestroyer so after i open the master file i go to where?
sorry im such a noob at this lol

DarthDestroyer, 2017-07-07:
top right window in VS is the file explorer.. navigate through it and open the files, they are listed in the changelog 

i will add the link, gimme  a second

jakobx13, 2017-07-07:
sweet thanks

jakobx13, 2017-07-07:
okay i think im starting to get it

jakobx13, 2017-07-07:
anything thats red in the link you gave me gets deleted

jakobx13, 2017-07-07:
anything green is added

jakobx13, 2017-07-07:
is that right

DarthDestroyer, 2017-07-07:
yes and no ;-)

red means old, green is new! 

The easiest (dont think) method is: delete red and add green

but, if you change only a dot, it will get marked red and the new line with the Addition gets green. 
wbat i mean is, you can change the old one to the new, if you see the changes ;-)

jakobx13, 2017-07-07:
okay

jakobx13, 2017-07-07:
so if its red in the link delete it in my file
if its green in the link add it in my file

jakobx13, 2017-07-07:
is it when words are red
or when the lines they're on are red

jakobx13, 2017-07-07:
i got it now
thanks man

DarthDestroyer, 2017-07-07:
it is a easy compare! 

like i said, red = old & green = new! 

if there is no red, it is only adding the green. 
and yes, it is per line

https://github.com/openhardwaremonitor/openhardwaremonitor/pull/884/files

jakobx13, 2017-07-07:
Sweet @DarthDestroyer ive done that
what now?

DarthDestroyer, 2017-07-07:
like i said, press "Run"

jakobx13, 2017-07-07:
in where? sorry

DarthDestroyer, 2017-07-07:
visual Studio, top panel

jakobx13, 2017-07-07:
also i've been hitting save on everything i have changed is that right?

DarthDestroyer, 2017-07-07:
of course

jakobx13, 2017-07-07:
well i get that but in what file

jakobx13, 2017-07-07:
like where should i be when i hit compile

DarthDestroyer, 2017-07-07:
save folder, bin, debug

if you switch to release, save folder, bin, release

jakobx13, 2017-07-07:
i dont see a bin file in the source code

DarthDestroyer, 2017-07-07:
You have to hit the green play Button in the top panel of VS! 
After that, it should start OHM with your addition! 
You can close that and the executable should be there!

jakobx13, 2017-07-07:
so just any file within the source file
hit green play button
see i get what you're saying but when i open the whole file with VS i get a blank screen

jakobx13, 2017-07-07:
okay so it brings up a selection of .exe files if i do it to the ohm.exe ittle work?

jakobx13, 2017-07-07:
sorry i am terible at this

DarthDestroyer, 2017-07-07:
i dont get, what you say :/

if you had changed all files - fine! 
save all and close
go to root folder of Sourcecode 
open ".csproj" file by doubleclick
there, in this opening Visual Studio Window, press Run
It should open the good old known OHM Window
Close that Window, close VS
In the save directory should now be a Bin Folder, there should be a Debug / Release Folder. In that, there is the requested executable!

jakobx13, 2017-07-07:
ohhhhhhhhhhh .csproj
i dont know if you have said that to me before

jakobx13, 2017-07-07:
now i have two of them i assume its the 2nd one correct?
OpenHardwareMonitorLib.csproj
OpenHardwareMonitor.csproj

ckasprzak, 2017-08-06:
When is the next update that will have AMD Ryzen Temp support compiled into it?

DarthDestroyer, 2017-08-06:
when its done ™ or soon ™

the nearest release is, when you compile it yourself! 
Sorry to say that...

ckasprzak, 2017-08-06:
:( Yeah I'm getting a VM setup for that I guess.