openhardwaremonitor/openhardwaremonitor #999: Intel Core i3-7100 - No temps

SearchForTheCode, 2017-07-06:
No temperature sensors found on the given cpu. 
I attached the report. Hopefully someone can find information out of this :)

[OpenHardwareMonitor.Report.txt](https://github.com/openhardwaremonitor/openhardwaremonitor/files/1127863/OpenHardwareMonitor.Report.txt)

DarthDestroyer, 2017-07-06:
http://openhardwaremonitor.org/files/openhardwaremonitor-v0.8.0.3-alpha.zip

That should help ;-)

SearchForTheCode, 2017-07-06:
Thats great..is this included in the latest sources you can download here? (I ask because i always need to modify the sources because of some problems with the backup software from symantec... and the generated .sys file)

DarthDestroyer, 2017-07-06:
it should be included, correct!

DarthDestroyer, 2017-07-07:
hui... maybe he missed to change the Version string PR

SearchForTheCode, 2017-07-07:
Can happen ;) Thank you, i will test the latest build then :)