openhardwaremonitor/openhardwaremonitor #1004: Add support for NCT6796D

cobrafast, 2017-07-24:
Hello, I'd like to add support for the Nuvoton NCT6796D as found on my ASUS Prime X299-Deluxe mainboard.

I have started implementing it based on other pull requests as a NCT6779/6791 chip.

This is how far I got:

[image]

As you can see the chip is detected, but none of its sensors are.
I simply added it to https://github.com/openhardwaremonitor/openhardwaremonitor/blob/master/Hardware/LPC/NCT677X.cs#L274 which I guess wasn't right.
However I don't feel comfortable implementing it as a new chip by itself as I have no idea where to get the required numbers and couldn't find any information about this particular chip online.

I'm also unfamiliar with the OHWM codebase and microcontroller programming in general.

If anyone could help me out and point me in the right direction that'd be great.

myfreeweb, 2017-07-24:
https://github.com/openhardwaremonitor/openhardwaremonitor/pull/884

I've added the NCT6795D https://github.com/openhardwaremonitor/openhardwaremonitor/pull/884/commits/6c0e00b8322d785eef65daaecb3dd0be5d60c328

You should try the same changes I did, with the correct ID for your chip of course

cobrafast, 2017-07-24:
Thanks that helped.
Cont in https://github.com/openhardwaremonitor/openhardwaremonitor/pull/1005#issuecomment-317565816

closed 2017-07-24 by cobrafast