openhardwaremonitor/openhardwaremonitor #1006: Multi GPU Support

rijincp, 2017-07-27:
Looks like multiple GPUs are not supported.
I'm using Nvidia Geforce 1070 Cards on ASUS maximus series motherboard. Not sure if everyone is having the same problem.

DarthDestroyer, 2017-07-27:
Could you provide a Report? That would help!

f-r-e-e-s-k-y, 2017-08-06:
I have multiple rigs ranging from intel 965, x39, p45, z87 to amd 890fx, each with multiple 1060 running, no issues here. All 1060´s with temps & usage info etc. are shown.

bigpodgurc, 2017-08-06:
The problem might be in sli

DarthDestroyer, 2017-08-06:
There shouldnt be an issue! 
But without a Report, we could only look into the crystal ball...

bigpodgurc, 2017-08-06:
difference between those mining rigs and rig of rijincp is SLI and it might do some different reporting or something. But im not 100% if its true we need another SLI rig to replicate

samadoni, 2017-10-18:
hi,
i can confirm that, i have 3 gpus although the temp and fan speed are correct but when it comes to memory we have exactly the same numbers which is wrong. (basicaly it reads only the mem data of the 1st gpu)

thanks   

[OpenHardwareMonitor2.Report.txt](https://github.com/openhardwaremonitor/openhardwaremonitor/files/1395332/OpenHardwareMonitor2.Report.txt)

rijincp, 2018-03-16:
Apologies for the late reply folks. I missed my comment notifications. Ended up not using the utility at the time. IIRC, I faced the same issue as @samadoni