openhardwaremonitor/openhardwaremonitor #1007: Using OHM in my own software

bigpodgurc, 2017-07-30:
I hvae school project and i wanna use OHM as a part or at least its api and i might change a thing or two in main software(that it starts and shutsdown automaticly when i open my program also i iwll make so it acutaly doestn show upanywheere but in processes. so who to tell what im gonna do and stuff like this

DarthDestroyer, 2017-07-30:
You dont have to change the Main for that! 

implement the lib and you get all infos you need, the start up and other things are possible in your application! 

but im sorry, that i wont do your homework or project! 
if you need help for that, give me your try and anyone will help! 

have fun with that awesome lib!

bigpodgurc, 2017-07-30:
i probably will just kill the taskicon of the OHM client and add something like a check if my software is not running it shuts down OHM client automaticly(lets say if my program crashes)

DarthDestroyer, 2017-07-30:
you have not understood me!

implement the "lib".dll into your project and you dont have to mess with the UI!
Every hardware Info is available there! Build your own tool without Taskicon and you are done

bigpodgurc, 2017-07-30:
im writing here because as far as i know you need to tell the owner of the project that you are changing the project. thats how i understood the licence

bigpodgurc, 2017-07-30:
didnt know that because when i try getting stuff like cpu load without OHM client running it doesnt get anything.

bigpodgurc, 2017-07-30:
thx im still learning stuff

DarthDestroyer, 2017-07-30:
okay, i give you some tips!

first, you dont have to "tell" anything about your project here! Only if you wanna sell your program / tool to anybody!

second, google for implementing dll into another project! It is well explained! No need for "a running OHM instance"

third, anyone had started small ;)

bigpodgurc, 2017-07-30:
Thx again and i was just going on safer side, better safe then sorry type a thing) because its school project