openhardwaremonitor/openhardwaremonitor #1008: Question lib and implementation of it

bigpodgurc, 2017-07-30:
`public static void CPULoadCollection()
            {
                Computer c = new Computer
                {
                    CPUEnabled = true
                };
                c.Open();

                foreach (var hardware in c.Hardware)
                {
                    if (hardware.HardwareType == HardwareType.CPU)
                    {
                        hardware.Update();
                        foreach (var sensors in hardware.Sensors)
                        {
                            if (sensors.SensorType == SensorType.Load)
                            {
                                if (sensors.Name == "CPU Total")
                                {
                                    DataCollecitonVar.CPULoad_Total = (int)sensors.Value.GetValueOrDefault();
                                }
                            }
                        }
                    }
                }
            }`

This is how i get the value ub CPU total but it takes hell of a long time to get it is there a better way

bigpodgurc, 2017-07-30:
basicly that code is realy slow specialy when you put together multiple diferent things at once(GPU, CPU...)

DarthDestroyer, 2017-07-30:
how often do you call CPULoadCollection?

There is no need to create a Computer everytime!

BTW: Stick at one Issue, please!

bigpodgurc, 2017-07-30:
thats the whole code i do to check for CPU load nothing more and this code is in a thread that also checks for gpu ram and couple of other stuff

bigpodgurc, 2017-07-30:
so is there any other better way

DarthDestroyer, 2017-07-30:
of course!

do that before the function

{
Computer c = new Computer
{
CPUEnabled = true
};
c.Open();

the rest of that function is okay!

bigpodgurc, 2017-07-30:
so if i want to collect for lets say CPU and GPU can i do 
Computer c = new Computer
{
CPUEnabled = true
GPUEnabled = true
};
c.Open();

or must i make separate for each

DarthDestroyer, 2017-07-30:
nope, that is correct!

bigpodgurc, 2017-07-31:
as it appears the program is outputing compleatly wrong numbers i have run this program and im getting as CPU Total Compleatly different number sometimes even twice as much as OHM shows sometime i even get 0