openhardwaremonitor/openhardwaremonitor #1010: Intel Core i7-7700HQ has no temperature

zhangfuyang, 2017-08-04:
I download version 0.8.0.0, and there is missing temperature sensors.
[image]

Also I want to use the API, and my code is below
```C#
                for (int i = 0; i < computer.Hardware.Length; i++)
                { 
                    if (computer.Hardware[i].HardwareType == HardwareType.CPU)
                    {
                        for (int j = 0; j < computer.Hardware[i].Sensors.Length; j++)
                        {
                            if (computer.Hardware[i].Sensors[j].SensorType == SensorType.Temperature)
                            {
                                if (computer.Hardware[i].Sensors[j].Name == "CPU Package") 
                                    Console.WriteLine("Tem=" + computer.Hardware[i].Sensors[j].Value.ToString());
                            }
                        }
                    }
                }
```
but I find "computer.Hardware[i].Sensors[j].SensorType" never equal to "SensorType.Temperature", I don't know how to fix it.
I am grateful if you could help me.

DarthDestroyer, 2017-08-11:
http://openhardwaremonitor.org/files/openhardwaremonitor-v0.8.0.3-alpha.zip

That should help ;-)
If not, could you provide a Report, please?

bocekm, 2018-02-09:
Works fine, thanks.
[image]

matrixluk, 2019-04-23:
> 
> 
> http://openhardwaremonitor.org/files/openhardwaremonitor-v0.8.0.3-alpha.zip
> 
> That should help ;-)
> If not, could you provide a Report, please?

Works very well! :D Thank You very much:)

okostyuk, 2019-07-03:
It doesnt work for me:
0 8 0-beta
[image]
0 8 3-alpha
[image]

DarthDestroyer, 2019-07-03:
@okostyuk

could you attach a report? 
it looks more like a linux thing...

okostyuk, 2019-07-08:
[OpenHardwareMonitor.Report.txt](https://github.com/openhardwaremonitor/openhardwaremonitor/files/3368613/OpenHardwareMonitor.Report.txt)

DarthDestroyer, 2019-07-08:
Look at #975 #867