openhardwaremonitor/openhardwaremonitor #1013: How does OHM actually get the hardware information?

iamanai, 2017-08-13:
I have been poking around OHM's code, trying to find out how it gets the sensor information from the computer hardware. This is mostly for my own curiosity, but it's also so I can potentially contribute to the project later on. I just can't get my head wrapped around how it works.

Could someone provide a quick explanation of how OHM gets its information? Thanks in advance.

DarthDestroyer, 2017-08-13:
There are several ways to query the hardware informations!

In the Hardware Directory you find all Sensors and its way to get there Data.
What do you search exactly?

iamanai, 2017-08-31:
Sorry for the late reply, I've been terribly busy. I have been looking through that folder, and I can only really tell that it somehow gets the info from the WinRing0 driver. The main reason I want to know is I want to try making my own small program (as a personal project) that will get basic information for a CPU or a GPU.

Tuupertunut, 2017-10-26:
I'm interested in this too. I can see the code in the Hardware folder but it's full of some hexadecimal numbers coming from a library called WinRing0, which itself is also a bunch of hexadecimals. I don't really understand how it works. Could someone explain in detail how OpenHardwareMonitor accesses the hardware sensors?

martynsmall, 2017-10-26:
I too was interested in this and read through a fair bit of the code to try and work it out.  The below is by no means complete, but I feel it gives enough of an overview that you can start investigating certain areas in more depth.

Different types of sensor are read in different ways, for example:
- CPU load, RAM and SMART come from windows API calls
- NVIDIA graphics cards come from NVIDIA dlls
- Mainboard temperatures/voltages/fans/etc and CPU temperatures/speeds/etc come from the WinRing0 driver (see below).

`WinRing0[x64].sys` is extracted and installed as a kernel-mode driver as this is the only way to directly access the registers/ports where the raw hardware values are found.  Basically this driver just provides a means for reading/writing port values, there's nothing specific to any types of hardware in it (the code can be found in the `External` folder of this repo).  `Ring0.cs` and `KernelDriver.cs` perform the extraction and installation.

The hexadecimal numbers you see throughout are largely port numbers of the locations where all the raw data is available.  These are obtained from the datasheets for the different I/O chips that are supported.

In a nutshell the application determines which I/O chip is present by reading the ID and revision from a couple of ports, then sets up sensors using the specific ports known to that chip.  `LPCIO.cs` does this detection and creation of the appropriate `ISuperIO` implementation.

As a final layer `SuperIOHardware.cs`  knows of particular motherboard models and creates the actual sensor items that you see on the UI.  Where a board isn't known it just uses a default configuration for the chip, hence the presence of many sensors such as `Voltage #1` or `Temperature #1`.  From what I gather this information is usually determined manually by either using the BIOS or running the manufacturer software that labels everything correctly and comparing the values with the ones from this application to establish which sensor maps to which.  An extra complication here is that many voltages are scaled, so the scaling factor must be known to reverse this and show the correct value.  This is again where the datasheet for the chip is required.