openhardwaremonitor/openhardwaremonitor #1014: Mainboard GigaByte H270N-WIFI-CF can't get temperature and fan speed CPU

elierodriguez, 2017-08-14:
Mainboard GigaByte H270N-WIFI-CF can't get temperature and fan speed CPU

this is the Report 

[OpenHardwareMonitor.Report.txt](https://github.com/openhardwaremonitor/openhardwaremonitor/files/1221957/OpenHardwareMonitor.Report.txt)

DarthDestroyer, 2017-08-14:
First, use the latest Version: http://openhardwaremonitor.org/files/openhardwaremonitor-v0.8.0.3-alpha.zip

Second, you have to run OHM as Admin!

That should solve your Issue!

elierodriguez, 2017-08-14:
The Version: 0.8.0.3

It doesn't show me the fan speed and the temperature either

this is the last Report
 Version: 0.8.0.3: 
[OpenHardwareMonitor.Report.txt](https://github.com/openhardwaremonitor/openhardwaremonitor/files/1222290/OpenHardwareMonitor.Report.txt)

DarthDestroyer, 2017-08-14:
Are you sure, that you had started as Admin?

elierodriguez, 2017-08-14:
this is the report again like admin.
[OpenHardwareMonitor.Report-links-admin.txt](https://github.com/openhardwaremonitor/openhardwaremonitor/files/1222385/OpenHardwareMonitor.Report-links-admin.txt)

elierodriguez, 2017-08-14:
I don't see what they fix to solve the problem

Do you have a sample or a better idea about how to fix it?

elierodriguez, 2017-08-16:
It work's...! Thanks...