openhardwaremonitor/openhardwaremonitor #1016: Gigabyte Z270X-Gaming 7 no voltages or fans speed info

Hosney74, 2017-08-16:
Mainboard Gigabyte Z270X-Gaming 7 can't get any information such as Voltages and CPU fan speed
this is the Report

[OpenHardwareMonitor.Report.txt](https://github.com/openhardwaremonitor/openhardwaremonitor/files/1227264/OpenHardwareMonitor.Report.txt)

Thanks

DarthDestroyer, 2017-08-16:
Should be the same Issue as #1014!

Hosney74, 2017-08-17:
sorry but i dont know waht to do to make it work
can you give me more help
thanks

Hosney74, 2017-09-12:
No one is gona help me with this?

DarthDestroyer, 2017-09-12:
i tried!

you have not much other options...
go to #1014, read it, follow it to the Pull Request #884 and do, what there is listed at file changes.

It isnt really hard and you will make the first step into coding ;-)

There are enough ppl who could do it for you, me included, but if i would be you... i wouldnt download a dll from an internet-guy :-P

You will see, how easy it is...