openhardwaremonitor/openhardwaremonitor #1022: List latest alpha/beta releases on http://openhardwaremonitor.org/ and GitHub release page

akej74, 2017-08-24:
Hi, hhanks for this great application, I have small enhancement request. I was looking for the latest alpha/beta release and found out by accident that e.g. http://openhardwaremonitor.org/files/openhardwaremonitor-v0.8.0.3-alpha.zip exists. This info is good byt is hidden away in some Issues on GitHub or comments on the official release for 0.8.0 beta on http://openhardwaremonitor.org/downloads/

My suggestion would be to add this to the official download page and also to the GitHub release page if possible.

RaydenX93, 2017-09-01:
i second this, good idea

Hubfront, 2017-09-10:
Good idea, i believe many would appreciate the possibility to download/test a more up-to-date unofficial binary.

rainkinz, 2017-11-19:
Where are the alpha releases? Are the on GH anywhere? Thanks