openhardwaremonitor/openhardwaremonitor #1028: Slightly inaccurate voltage

end360, 2017-09-06:
GPU: AMD R7 200 series
Problem: http://imgur.com/a/vlVWi

colorburst, 2017-09-22:
Please post the report for us to look at, it might shed light on the issue.

end360, 2017-09-22:
I will as soon as I get home ( about 6 hours )

DarthDestroyer, 2017-09-22:
I think i had seen another Issue like that...
#825 and a old one #447

825 talks about Voltage, but maybe it is related?!

A Report would help here, yes!

end360, 2017-09-22:
[OpenHardwareMonitor.Report.txt](https://github.com/openhardwaremonitor/openhardwaremonitor/files/1326086/OpenHardwareMonitor.Report.txt)

DarthDestroyer, 2017-09-22:
could you try the latest version, too?

http://openhardwaremonitor.org/files/openhardwaremonitor-v0.8.0.3-alpha.zip

end360, 2017-09-23:
Still the same: 
[OpenHardwareMonitor.Report2.txt](https://github.com/openhardwaremonitor/openhardwaremonitor/files/1326148/OpenHardwareMonitor.Report2.txt)

closed 2020-01-31 by end360