openhardwaremonitor/openhardwaremonitor #1029: library not retrieving all voltages

Joni-Salminen, 2017-09-09:
Hi, i stumbled upon issue of not getting 3.3v, 5v and 12v voltages from the .DLL file(self compiled to add my motherboard in there)

So the question is, am i even supposed to get the power supply line values or do i need to look for those elsewhere? SuperIO hardware return float[15] for the voltages, which does not include +3.3, +5v and +12v.

colorburst, 2017-09-22:
It depends on your motherboard and SuperIO chip -- there is no standard, so each manufacturer measures voltages differently, if at all.

Joni-Salminen, 2017-09-22:
Uh, I actually forgot to close and update this, I got the voltages, I just had to 5x one of the value for 5v line and 12x for 12v, which completely makes sense since the chip on motherboard propably is 3.3v powered and thus cannot measure higher than that, so there is voltage divider on the ADC input,  so I need to scale it back up on software.

colorburst, 2017-09-22:
How did you verify that the voltages actually correspond to the correct power rails? You could multiply any measurement in the unit range by 5 or 12 and get a plausible reading.

Joni-Salminen, 2017-09-23:
I had Asus AI suite open with my application and I used voltages it does show as reference, I basically changed all the voltages little bit from the Asus software to the point i only had 2 uknown values left on my application, which I then did find out by loading the PSU to the point i can see drift in the power lines to figure which is 5v and 12v.

colorburst, 2017-09-23:
I see, very nice! If you want, you can post your modifications here or make a pull request, for the benefit of other people who have the same motherboard.

Joni-Salminen, 2017-09-24:
I will make pull request once i have time to figure all the temperatures.