openhardwaremonitor/openhardwaremonitor #1031: How to reset everything

stl3, 2017-09-16:
Hi, I am having an issue after I hid a few items in mainboard temperature settings. The reason I did that is because out of the blue OWM started showing weird values and an extra core temp. Now I basically want to reset everything but even removing OpenHardwareMonitor.config doesn't reset all of it, I cannot get Temperature #1 and #4 back so I assume there is somewhere in the registry that is keeping these values that I can delete to unhide them again? Is there a simpler way to unhide what I have hidden? I've used both 0.8.0.0 and 0.8.0.3 and I just cannot get it back to default. Mainboard only shows CPU Core, and temperatures #2, #3 and #5

stl3, 2017-09-16:
Open Hardware Monitor Report

--------------------------------------------------------------------------------

Version: 0.8.0.0

--------------------------------------------------------------------------------

Common Language Runtime: 4.0.30319.42000
Operating System: Microsoft Windows NT 6.1.7601 Service Pack 1
Process Type: 64-Bit

--------------------------------------------------------------------------------

Sensors

|
+- ASUS H97-PRO GAMER (/mainboard)
|  |
|  +- Nuvoton NCT6791D (/lpc/nct6791d)
|  |  +- CPU VCore      :    0.904    0.904    0.904 (/lpc/nct6791d/voltage/0)
|  |  +- Voltage #2     :    1.024    1.024    1.024 (/lpc/nct6791d/voltage/1)
|  |  +- AVCC           :     3.36     3.36    3.376 (/lpc/nct6791d/voltage/2)
|  |  +- 3VCC           :     3.36     3.36     3.36 (/lpc/nct6791d/voltage/3)
|  |  +- Voltage #5     :    1.024    1.016    1.024 (/lpc/nct6791d/voltage/4)
|  |  +- Voltage #6     :    2.032    2.024    2.032 (/lpc/nct6791d/voltage/5)
|  |  +- Voltage #7     :    0.728    0.728    0.728 (/lpc/nct6791d/voltage/6)
|  |  +- 3VSB           :     3.44     3.44     3.44 (/lpc/nct6791d/voltage/7)
|  |  +- VTT            :    1.008    1.008    1.016 (/lpc/nct6791d/voltage/9)
|  |  +- Voltage #11    :     2.04     2.04     2.04 (/lpc/nct6791d/voltage/10)
|  |  +- Voltage #12    :    1.008    1.008    1.008 (/lpc/nct6791d/voltage/11)
|  |  +- Voltage #13    :    0.624    0.624    0.624 (/lpc/nct6791d/voltage/12)
|  |  +- Voltage #14    :    0.616    0.616    0.616 (/lpc/nct6791d/voltage/13)
|  |  +- CPU Core       :     31.5       29       35 (/lpc/nct6791d/temperature/0)
|  |  +- Temperature #2 :       32       32       32 (/lpc/nct6791d/temperature/2)
|  |  +- Temperature #3 :       41       41       41 (/lpc/nct6791d/temperature/3)
|  |  +- Temperature #5 :       25       25       26 (/lpc/nct6791d/temperature/5)
|  |  +- Fan #2         :     1535     1478     1535 (/lpc/nct6791d/fan/1)
|  |  +- Fan #6         :     1122     1117     1135 (/lpc/nct6791d/fan/5)
|  |  +- Fan Control #1 :      100      100      100 (/lpc/nct6791d/control/0)
|  |  +- Fan Control #2 :      100      100      100 (/lpc/nct6791d/control/1)
|  |  +- Fan Control #3 :      100      100      100 (/lpc/nct6791d/control/2)
|  |  +- Fan Control #4 :      100      100      100 (/lpc/nct6791d/control/3)
|  |  +- Fan Control #5 :      100      100      100 (/lpc/nct6791d/control/4)
|  |  +- Fan Control #6 :      100      100      100 (/lpc/nct6791d/control/5)
|
+- Intel Core i7-4790K (/intelcpu/0)
|  +- Bus Speed      :  100.001  100.001  100.008 (/intelcpu/0/clock/0)
|  +- CPU Core #1    :  2400.02  800.005  4500.37 (/intelcpu/0/clock/1)
|  +- CPU Core #2    :  2400.02  800.005  4500.04 (/intelcpu/0/clock/2)
|  +- CPU Core #3    :  2400.02  800.005  4500.04 (/intelcpu/0/clock/3)
|  +- CPU Core #4    :  2400.02  800.005  4500.04 (/intelcpu/0/clock/4)
|  +- CPU Core #1    :       40       38       47 (/intelcpu/0/temperature/0)
|  +- CPU Core #2    :       42       40       53 (/intelcpu/0/temperature/1)
|  +- CPU Core #3    :       38       36       50 (/intelcpu/0/temperature/2)
|  +- CPU Core #4    :       39       36       46 (/intelcpu/0/temperature/3)
|  +- CPU Package    :       42       40       53 (/intelcpu/0/temperature/4)
|  +- CPU Total      :  13.4615  8.84615  18.1818 (/intelcpu/0/load/0)
|  +- CPU Core #1    :  16.1539  8.46154  36.3636 (/intelcpu/0/load/1)
|  +- CPU Core #2    :  12.3077  4.54545  19.7674 (/intelcpu/0/load/2)
|  +- CPU Core #3    :  12.3077  9.23077  18.6047 (/intelcpu/0/load/3)
|  +- CPU Core #4    :  13.0769  4.61538  20.3488 (/intelcpu/0/load/4)
|  +- CPU Package    :  25.7808  17.1994   36.292 (/intelcpu/0/power/0)
|  +- CPU Cores      :  17.3114  9.05598  29.2207 (/intelcpu/0/power/1)
|  +- CPU Graphics   :        0        0        0 (/intelcpu/0/power/2)
|  +- CPU DRAM       :  3.25243  2.96133  3.64357 (/intelcpu/0/power/3)
|
+- Generic Memory (/ram)
|  +- Memory         :  52.5144  52.4523  52.7204 (/ram/load/0)
|  +- Used Memory    :    8.373  8.36311  8.40586 (/ram/data/0)
[... 982 lines cut ...]
Firmware version: 80.00A80

 ID Description                        Raw Value    Worst Value Thres Physical
 01 Read Error Rate                    000000000000 200   200   51    -       
 03 Spin-Up Time                       1D1000000000 172   177   21    -       
 04 Start/Stop Count                   821F00000000 92    92    0     8066    
 05 Reallocated Sectors Count          000000000000 200   200   140   -       
 07 Seek Error Rate                    000000000000 200   200   0     -       
 09 Power-On Hours (POH)               8B2900000000 86    86    0     10635   
 0A Spin Retry Count                   000000000000 100   100   0     -       
 0B Recalibration Retries              000000000000 100   100   0     -       
 0C Power Cycle Count                  D60000000000 100   100   0     214     
 C0 Emergency Retract Cycle Count      8C0000000000 200   200   0     -       
 C1 Load Cycle Count                   863402000000 152   152   0     -       
 C2 Temperature                        230000000000 102   112   0     35      
 C4 Reallocation Event Count           000000000000 200   200   0     -       
 C5 Current Pending Sector Count       000000000000 200   200   0     -       
 C6 Uncorrectable Sector Count         000000000000 200   200   0     -       
 C7 UltraDMA CRC Error Count           000000000000 200   200   0     -       
 C8 Write Error Rate                   000000000000 200   200   0     -       

Logical drive name: F:\
Format: NTFS
Total size: 2000388063232
Total free space: 249799368704

stl3, 2017-09-16:
Temperature #1 just popped up but it's showing -47.5C and max of -14.5C
Could this be due to heatsink needing reseating? I changed my heatsink/cooler recently as well, but not sure if they are related. For peace of mind I will redo the paste/reseat and see what happens from there.

ArmandoRl1, 2021-02-21:
Did you find a solution to this?

stl3, 2021-02-21:
> 
> 
> Found a solution to this: In the View Tab there's a "Show Hidden Sensors" option that lets you see and unhide whatever you hid.

I did try that before but it didn't help with my problem at the time. I will check if anything has changed in v0.9.6

stl3, 2021-02-21:
If you are curious what it shows now, this is the generated OpenHardwareMonitor.Report.txt for v0.9.6
No changes to hardware (that I remember anyway).

[OpenHardwareMonitor.Report.txt](https://github.com/openhardwaremonitor/openhardwaremonitor/files/6017632/OpenHardwareMonitor.Report.txt)