openhardwaremonitor/openhardwaremonitor #1036: External fan control - adding a "setter" method to WMI implementation of Control instances?

cosmicdan, 2017-10-02:
Title is self-explanatory.

It would be great if we had a method to set the values for e.g. fan speed, probably through WMI would be the "correct" way. I took a crack at this, but this is the first time I've ever worked with WMI - it seems we need the [ManagementTask attribute](https://msdn.microsoft.com/en-us/library/system.management.instrumentation.managementtaskattribute(v=vs.90).aspx) (on a "setter" method in Sensor.cs) which only appears in .NET 3.5 onwards, and - ignoring potential breakage with Linux hosts - updating the project to .NET 3.5 produced some weird .NET error about ambiguity that confuses the heck out of me.

Any ideas? Would love to be able to defer fan control to my own implementation - I can't think of any other way.

DarthDestroyer, 2017-10-03:
I would ever prefer the direct dll (lib) Implementation before i would use wmi!

I dont know, what you do or want... But if you use .net i would use the way i had said!

the wmi Part could need some love, sooner or later ;-)

cosmicdan, 2017-10-04:
Well that's not really how DllCall's are supposed to be used. A dllcall could be implemented as a last resort or alternative, but it is certainly a poor substitute for using COM and the WMI API which works great - it's much more flexible and language-agnostic.

My issue is specifically about giving OHM controller objects read/write values rather than just read-only values so other programs and scripts (e.g.  PowerShell) can automate hardware monitoring and fan control themselves. The issue here is that to make a writable value in WMI you need to move to .NET 3.5, and in my initial investigation OxyPlot has issues building under 3.5 and I have no idea about Linux support for that (Mono or whatever).

j3hyde, 2018-06-17:
This weekend I took a dive into this same topic.  I too found that I had to upgrade .NET Framework to get writable properties, methods, etc.  To do that I also upgraded OxyPlot as it now targets .NET 4.  See PR #1111.

I then implemented a WMI class called `Control` for things like fans. That's in PR #1112.  I can now set fan speed via WMI for example.  As this may break compatibility for Linux it may be better to implement read/write via the JSON provider using a POST message.