openhardwaremonitor/openhardwaremonitor #1037: Reading AMD's Ryzen Tcl/TDie

BrianLima, 2017-10-04:
I'm using the fork by myfreewb and dernerv and it does support Ryzen motherboards, their fan curve support is really cool too, but i'm getting mixed results, HWiNFO reports higher temperatures than OHM because of Ryzen's Tcl/Tdie readings, would it be possible to have it on OHM?

Kraust, 2017-11-13:
[image]

Currently working on a solution for this. Stay Tuned. Took the k10temp implementation and ported it to OpenHardwareMonitor.

BrianLima, 2017-11-13:
I think that the 20c offset problem for X series cpu's is gone, they removed it on a bios update if i remember correctly.
 I own a 1700 too, thanks for the update! Just hope they comeback to merge all the changes the community made or that another fork takes its place.

FireFrenzy, 2017-12-18:
My brand new 1950x has the offset between Tdie and Tcl... 

And thats on a week old motherboard with the latest bios on it (gigabyte gaming 7) where there is also no OHM support...

moellerm, 2020-05-12:
This should be fixed in OHM version 0.9.0 and later.

closed 2020-05-12 by moellerm