openhardwaremonitor/openhardwaremonitor #1040: Gigabyte Z270M-D3H-CF

wanccie, 2017-10-21:
Any chance somebody can add support for this gigabyte mother board.  Gigabyte Z270M-D3H-CF

Thanks in advance.

DarthDestroyer, 2017-10-21:
There is already a PR in the Pipeline #884
If you want it now, there is no other way as implementing this PR into the Sourcecode by yourself! Sorry... 

But, to make sure that your Mainboard has the same Chipset which the PR handles, we could need a Report here! If you could upload one, that would be great!

Regards

wanccie, 2017-10-22:
Hey Thanks.... Not sure what you wanted but ran CPUID on my board and
deleted all of the crap you don't need but the basics.

Thanks again.  If this is not what you wanted just let me know and I will
try again.

LloydPrindle

CPU-Z TXT Report
-------------------------------------------------------------------------

Binaries
-------------------------------------------------------------------------

CPU-Z version			1.81.1.x64

Processors
-------------------------------------------------------------------------

Number of sockets		1
Number of threads		8

APICs
-------------------------------------------------------------------------

Socket 0	
	-- Core 0	
		-- Thread 0	0
		-- Thread 1	1
	-- Core 1	
		-- Thread 0	2
		-- Thread 1	3
	-- Core 2	
		-- Thread 0	4
		-- Thread 1	5
	-- Core 3	
		-- Thread 0	6
		-- Thread 1	7

Timers
-------------------------------------------------------------------------

	ACPI timer		3.580 MHz
	Perf timer		4.102 MHz
	Sys timer		1.000 KHz

Processors Information
-------------------------------------------------------------------------

Socket 1			ID = 0
	Number of cores		4 (max 4)
	Number of threads	8 (max 8)
	Name			Intel Core i7 7700K
	Codename		Kaby Lake
	Specification		Intel(R) Core(TM) i7-7700K CPU @ 4.20GHz
	Package (platform ID)	Socket 1151 LGA (0x1)
	CPUID			6.E.9
	Extended CPUID		6.9E
	Core Stepping		B0
	Technology		14 nm
	TDP Limit		91.0 Watts
	Tjmax			100.0 °C
	Core Speed		800.0 MHz
	Multiplier x Bus Speed	8.0 x 100.0 MHz
	Stock frequency		4200 MHz
	Max frequency		4600 MHz
	Instructions sets	MMX, SSE, SSE2, SSE3, SSSE3, SSE4.1, SSE4.2, EM64T, VT-x, AES, AVX, AVX2, FMA3, TSX
	L1 Data cache		4 x 32 KBytes, 8-way set associative, 64-byte line size
	L1 Instruction cache	4 x 32 KBytes, 8-way set associative, 64-byte line size
	L2 cache		4 x 256 KBytes, 4-way set associative, 64-byte line size
[... 154 lines cut ...]
	60	00 14 41 46 1C 01 00 C0 00 14 3C 46 1C 01 00 FF 
	70	00 14 3C 46 1C 01 00 2E 00 14 3C 46 1C 01 00 D0 
	80	30 03 00 00 FF FF FF FF 02 30 01 02 01 C1 E0 51 
	90	00 00 00 00 FF 10 10 10 40 97 00 00 40 05 C1 00 
	A0	00 00 96 00 00 0A 0F 98 7F 7F 96 80 00 00 0F 98 
	B0	BD D4 88 7F 7F 7F 89 00 7F 7F 3C 3C 3C 3C 3C 3C 
	C0	FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF 
	D0	FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF 01 
	E0	FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF 
	F0	FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF 

Hardware monitor		NVIDIA NVAPI
	Voltage 0		0.63 Volts [0x271] (GPU)
	Power 0			5.23 pc (GPU)
	Temperature 0		34 degC (93 degF) [0x22] (GPU)
	Fan 0			1070 RPM [0x42E] (GPU)
	Fan PWM 0		24 pc [0x18] (FANPWMIN0)
	Clock Speed 0		139.00 MHz [0x8B] (Graphics)
	Clock Speed 1		405.00 MHz [0x8B] (Memory)

Software
-------------------------------------------------------------------------

Windows Version			Microsoft Windows 10 (10.0) Home 64-bit   (Build 15063) 
DirectX Version			12.0

colorburst, 2017-10-22:
For IT8686, you want to look into #970.

DarthDestroyer, 2017-10-22:
You can create a Report with OHM through the Top Menu! 
I think it was File -> Create Report...

You will get a txt file, that you can drag and drop here...

wanccie, 2017-10-22:
Thanks....  
[OpenHardwareMonitor.Report.txt](https://github.com/openhardwaremonitor/openhardwaremonitor/files/1404930/OpenHardwareMonitor.Report.txt)

DarthDestroyer, 2017-10-22:
Chip ID: Unknown ITE with ID 0x8686 at 0x2E/0x2F

#970 will help you, @colorburst was correct! 
There is also an instruction on how you could do that, in that PR! 

by the way, you use an outdated Version ;-)
8.0.3 would support your cpu!

wanccie, 2017-10-22:
Daniel,

Sorry... I'm such an idiot.  You don't make it easy to find the 8.03
version, still my mistake.  Great job and you are correct it works if I use
the latest version.  HA....

Thanks again.
lp