openhardwaremonitor/openhardwaremonitor #1042: fan speed not showing

AndrejTelle, 2017-11-01:
Hi,

I have a acer aspire VN7-793G-52XN notebook. The fans are running without load and I want to check if they are running within specifications or whether there is some bug.

However, open hardware monitor does not display the fan speeds (as seen on some screenshots in the documentation/website):
[image]

Is there any way to turn this on?

Thank you,
Andrej

DarthDestroyer, 2017-11-03:
okay ;)

3 points:

1. could you upload a report from ohm? 
2. you should use the latest Version! 
3. your Mainboard isnt supported yet and a report could help here, too

DarthDestroyer, 2017-11-03:
http://openhardwaremonitor.org/files/openhardwaremonitor-v0.8.0.3-alpha.zip

csujedihy, 2017-11-06:
Glad to see this is still under development. The version on the official website cannot display a lot of information like CPU temp on my i7-7700 but this one works!

AndrejTelle, 2017-11-08:
I installed the update (0.8.0.3 alpha) and now I see the temperatures:
[image]
however still no fan speed

here is the report:
[OpenHardwareMonitor.Report.txt](https://github.com/openhardwaremonitor/openhardwaremonitor/files/1454761/OpenHardwareMonitor.Report.txt)

thanks again for helping :)

DarthDestroyer, 2018-06-05:
@sammcgrail That doesnt really help! Without any Information about your Rig, it is impossible to help you!

Upload a Report, and we can check how to solve your Issue!

sammcgrail, 2018-06-05:
[OpenHardwareMonitor.Report.txt](https://github.com/openhardwaremonitor/openhardwaremonitor/files/2073580/OpenHardwareMonitor.Report.txt)

Here

[image]

DarthDestroyer, 2018-06-05:
@sammcgrail You should have seen, what is written here!
First Mistake is: Not using latest Version => http://openhardwaremonitor.org/files/openhardwaremonitor-v0.8.0.3-alpha.zip

Second one: Chip ID: Unknown Winbond / Nuvoton / Fintek with ID 0xD423 at 0x2E/0x2F
Which is for now, a unknown Mainboard Chipset...

But you can try #981 or #884 and post your Info if it works, here!

dcheglov, 2019-12-12:
Hello,
I have an issue to get information from internal fan on Fujitsu computers using latest version of OHM tool. Also I have tested latest OHM source code v.8.0.5.0 Alpha. The information from internal fan does not present under UI. 
I tried to debug OHM source code and found exception error when prog
Duplicate of #ram trying to get information about fan, by calling “Add(new TBalancer.TBalancerGroup(settings));”. The Exception code received from "catch (DllNotFoundException) { return; }" “Unable to load DLL “ftd2xx.dll. The specified module could not be found. Exception from HRESULT:0x8007007E”. The screenshot is attached.
Also, OHM log file is attached.
Could you please help me to fix this problem?
[T938_OpenHardwareMonitor.Report.txt](https://github.com/openhardwaremonitor/openhardwaremonitor/files/3953274/T938_OpenHardwareMonitor.Report.txt)
[LIFEBOOK_T937_OpenHardwareMonitor.Report.txt](https://github.com/openhardwaremonitor/openhardwaremonitor/files/3953275/LIFEBOOK_T937_OpenHardwareMonitor.Report.txt)

[image]
#

Mountaingal, 2020-01-26:
Is this still being developed? I am using the latest version of OHM, but also have the problem that there are NO fans showing up. 
Below are a screen shot and and report
[image]
[OpenHardwareMonitor.Report.txt](https://github.com/openhardwaremonitor/openhardwaremonitor/files/4113809/OpenHardwareMonitor.Report.txt)

scoob8000, 2020-02-08:
Me too.   This is a Dell 3020 SFF system.    Speedfan is able to see the fan sensors, if I "Enable Dell Support" so I figure the sensors are there and exposed.   Just a matter of adding them.    My report is attached.   Thanks!
[LibreHardwareMonitor.Report.txt](https://github.com/openhardwaremonitor/openhardwaremonitor/files/4174184/LibreHardwareMonitor.Report.txt)

OverShifted, 2020-03-17:
I have same thing on Asus k570ud-f
use this: http://www.almico.com/sfdownload.php

FlRENADO, 2020-03-21:
I have the same problem on a Samsung laptop from 2012 but I was wondering if the problem could be that the mother board doesn't support external software having control of the fans?

HypurrFPV, 2021-06-12:
Same problem with an ASUS Prine-X470-Pro motherboard.  If I exit and restart OHM it will eventually see the fans but I may have to do that 2 or 3 times or more before it does.  It's as if it's not waiting long enough for a response from the sensors before it gives up trying to read them.