openhardwaremonitor/openhardwaremonitor #1043: GPU Reports Invalid Voltage on RX 400 series with newer AMD Drivers

Kraust, 2017-11-13:
Note: This might be a flaw in the ADL SDK. If I could create a post on AMD's Support Forum I would.

I've run through the sample ADL SDK projects and ADL2_OverdriveN does not report out the valid VDDC values as well. OpenHardwareMonitor still uses ADL_Overdrive5, which is reports out the same values -

Oddly enough the demo ADL_Overdrive5 code reports "Driver Unsupported" (Which is correct functionality - AFAIK Wattman Supports OverdriveN). 

Don't know if this is resolvable right now or if this is an AMD issue.

robertarnesson, 2017-12-22:
Also interested in this. HWInfo had the same issue but solved it somehow. I also have the vega 56, getting 0 VDDC with ADL and openhardwaremonitor

DarthDestroyer, 2017-12-22:
Could this #1028 help here?