openhardwaremonitor/openhardwaremonitor #1050: ASRock Z270 Taichi - Temperature #4 always appears high (105-107 deg C)

haroneyn, 2017-12-06:
Temperature #4 always appears very high, even at low load (105-107 deg C). All other temps appear normal at appx 20-30 deg C.

OHM displays CPU Core temp & Temperatures #1 through #6 (with #4 being the oddball / high result).

The CPUID HWMonitor app displays Package Temperature & Core #0 through #4 (all showing good temps in the 20s & 30s deg C - same with the other motherboard temps).

I don't know whether this Temperature #4 reading is bogus or not. I'm currently assuming so, but I need to research how these two applications get their results, as well as the differences in what they display.

Can someone tell me exactly what "Temperature #4" is monitoring?
See link below for Hardware Report export.
Thanks
Nic
haroneyn@hushmail.com

[12-06-17 348pm OpenHardwareMonitor.Report.txt](https://github.com/openhardwaremonitor/openhardwaremonitor/files/1536880/12-06-17.348pm.OpenHardwareMonitor.Report.txt)

DarthDestroyer, 2017-12-06:
try the latest Version, atleast for your cpu http://openhardwaremonitor.org/files/openhardwaremonitor-v0.8.0.3-alpha.zip

You are only reading Mainboard Sensors and these can be everywhere ;)

haroneyn, 2017-12-06:
Thanks for the quick reply! Ill check out the latest version.
Take care
Nic

Arffeh, 2017-12-23:
Hi, I have a 7820x on a tiachi, I see 6 of the 8 cores, and have the same bizarre 130c core sitting in the middle of the rest.
Is there anything I can provide to help hunt down the issue?

DarthDestroyer, 2017-12-23:
@Arffeh a Report would help! Latest Version used?