openhardwaremonitor/openhardwaremonitor #1053: Intel GPU temperature not reported

endmarsfr, 2017-12-16:
hello, 
First of all, I would like to congratulate you for this great job.

I just want to report you a little issue about no retrieving  of integrated Intel GPU temperature.

I have another feature request that could be also appreciated.
It could be great if i can connect to a remote computer (using credentials)  to monitor its sensor values.
I tested  web server feature, but it requires to let the application always running on the remote computer.
I prefer to connect to the distant machine only when i need it.

regards

DarthDestroyer, 2017-12-16:
afaik, intel isnt supported yet
(you could provide us a Report) 

for your remote request, the application has to run... 
there is no way around that! 
Another way could be the remote wmi, that you could try
but, as i said before, the application has to run!

endmarsfr, 2017-12-16:
Thank you for your answer.

for the remote request, I know the application has to run, but I mean : not all the time :)
I don't want to connect to the remote machine all the time because it is a mediacenter always playing.
Yes an option to connect to remote using WMI could be satisfaying :)

regards

closed 2019-03-23 by endmarsfr