openhardwaremonitor/openhardwaremonitor #1055: Intel Celeron 430 temperatyre not supported

FedyaLutkovski, 2017-12-18:
The temperature display is not supported on this old processor. Can I do something about this, or is this equipment not going to be supported?

DarthDestroyer, 2017-12-18:
Without a Report, it is impossible to say!

Could you provide one?

FedyaLutkovski, 2017-12-18:
Oh sure

Open Hardware Monitor Report

--------------------------------------------------------------------------------

Version: 0.8.0.3

--------------------------------------------------------------------------------

Common Language Runtime: 2.0.50727.8762
Operating System: Microsoft Windows NT 6.1.7601 Service Pack 1
Process Type: 32-Bit

--------------------------------------------------------------------------------

Sensors

|
+- ASUS P5KPL-AM (/mainboard)
|  |
|  +- Winbond W83627DHG (/lpc/w83627dhg)
|  |  +- CPU VCore      :     1.32     1.32     1.32 (/lpc/w83627dhg/voltage/0)
|  |  +- Voltage #2     :     1.84     1.84     1.84 (/lpc/w83627dhg/voltage/1)
|  |  +- AVCC           :    3.248    3.248    3.248 (/lpc/w83627dhg/voltage/2)
|  |  +- 3VCC           :    3.248    3.248    3.248 (/lpc/w83627dhg/voltage/3)
|  |  +- Voltage #5     :    1.256    1.256    1.256 (/lpc/w83627dhg/voltage/4)
|  |  +- Voltage #6     :    1.592    1.592    1.592 (/lpc/w83627dhg/voltage/5)
|  |  +- Voltage #7     :    1.224    1.224    1.224 (/lpc/w83627dhg/voltage/6)
|  |  +- 3VSB           :    3.248    3.248    3.248 (/lpc/w83627dhg/voltage/7)
|  |  +- Auxiliary      :       27       27       27 (/lpc/w83627dhg/temperature/1)
|  |  +- System         :       28       28       28 (/lpc/w83627dhg/temperature/2)
|  |  +- CPU Fan        :  1997.04  1997.04  1997.04 (/lpc/w83627dhg/fan/1)
|  |  +- Auxiliary Fan  :  1767.02  1767.02  1767.02 (/lpc/w83627dhg/fan/2)
|
+- Intel Celeron 430 (/intelcpu/0)
|  +- CPU Core       :  93.9655  93.9655      100 (/intelcpu/0/load/1)
|
+- Generic Memory (/ram)
|  +- Memory         :  53.8497  53.8497  53.8497 (/ram/load/0)
|  +- Used Memory    : 0.533333 0.533333 0.533333 (/ram/data/0)
|  +- Available Memory : 0.457077 0.457077 0.457077 (/ram/data/1)
|
+- ST380815AS (/hdd/0)
|  +- Temperature    :       34       34       34 (/hdd/0/temperature/0)
|  +- Used Space     :  80.3421  80.3421  80.3421 (/hdd/0/load/0)

--------------------------------------------------------------------------------

Parameters

|
+- ASUS P5KPL-AM (/mainboard)
|  |
|  +- Winbond W83627DHG (/lpc/w83627dhg)
|  |  |
|  |  +- CPU VCore (/lpc/w83627dhg/voltage/0)
|  |  |  +- Ri [k?] : 0 : 0
|  |  |  +- Rf [k?] : 1 : 1
|  |  |  +- Vf [V] : 0 : 0
|  |  |
|  |  +- Voltage #2 (/lpc/w83627dhg/voltage/1)
|  |  |  +- Ri [k?] : 0 : 0
|  |  |  +- Rf [k?] : 1 : 1
|  |  |  +- Vf [V] : 0 : 0
|  |  |
|  |  +- AVCC (/lpc/w83627dhg/voltage/2)
|  |  |  +- Ri [k?] : 34 : 34
|  |  |  +- Rf [k?] : 34 : 34
|  |  |  +- Vf [V] : 0 : 0
[... 258 lines cut ...]
 05 Reallocated Sectors Count          000000000000 100   100   36    -
 07 Seek Error Rate                    A2CFBE270000 60    87    30    -
 09 Power-On Hours (POH)               E85F00000000 72    72    0     24552
 0A Spin Retry Count                   000000000000 100   100   97    -
 0C Power Cycle Count                  5D0900000000 98    98    20    2397
 BB Reported Uncorrectable Errors      000000000000 100   100   0     0
 BD High Fly Writes                    000000000000 100   100   0     -
 BE Temperature Difference from 100    220017230000 54    66    45    34
 C2 Temperature                        220000001100 46    34    0     34
 C3 Hardware ECC Recovered             E8EE9C0D0000 69    74    0     -
 C5 Current Pending Sector Count       000000000000 100   100   0     -
 C6 Uncorrectable Sector Count         000000000000 100   100   0     -
 C7 UltraDMA CRC Error Count           000000000000 200   200   0     -
 C8 Write Error Rate                   000000000000 253   100   0     -
 CA Data Address Mark errors           000000000000 253   100   0     -

Logical drive name: C:\
Format: NTFS
Total size: 40541089792
Total free space: 6643462144

Logical drive name: D:\
Format: NTFS
Total size: 1734340608
Total free space: 1666990080

DarthDestroyer, 2017-12-22:
I dont know, that this cpu is supported or not - by Intel or the Way, how OHM gets the Data!
You have to understand, that your CPU is 10 Years old!

Atleast, you get Mainboard Data ;)

Intel Ark https://ark.intel.com/products/29735/Intel-Celeron-Processor-430-512K-Cache-1_80-GHz-800-MHz-FSB

closed 2018-01-04 by FedyaLutkovski