openhardwaremonitor/openhardwaremonitor #1059: SSL support

VahaC, 2017-12-22:
How about SSL certificates support?

DarthDestroyer, 2017-12-22:
#964

VahaC, 2017-12-22:
I'm talking not about site but about app

DarthDestroyer, 2017-12-22:
lol, why arent you saying that you mean "digitally signing"?

There are several Reasons, why not!
1. To sign an Application, you need a Company to "vouch" you - that costs Money (Verisign, Thawte and so on)
2. You could sign it by yourself for free
3. It is not needed!

Let me combine all these Reasons into one Answer:
OHM is Open-Source! You can view the complete Code, can look what it does and why.
There is in no Part, a payed Service needed to "secure" it!
If OHM got self signed, who says that it's secure?! 

I hope you get the Point!

closed 2017-12-22 by VahaC