openhardwaremonitor/openhardwaremonitor #1060: Fans control in Maximus X Hero (z370)

xXemix, 2017-12-25:
Hello everyone, and Merry Christmas! First of all forgive my English, since I am Spanish and there are phrases that I do not know very well.
I wanted to ask you if it is possible to add the control of the fans to this wonderful application that I have been using with all my previous motherboards but now I have changed to the Maximus X Hero and it does not detect the fans.

I leave a report and any help or question, do it without problems.

Regards!
[OpenHardwareMonitor.Report.txt](https://github.com/openhardwaremonitor/openhardwaremonitor/files/1585820/OpenHardwareMonitor.Report.txt)

DarthDestroyer, 2017-12-25:
#981

Chip ID: Unknown Winbond / Nuvoton / Fintek with ID 0xD121 at 0x2E/0x2F

xXemix, 2017-12-25:
Perfect! I just compiled the new lines of code of the post that you have indicated and everything perfect! I'm back in control of the fans!

Many thanks and happy Christmas again!

DarthDestroyer, 2017-12-25:
Glad, that i helped!

Happy Xmas, too