openhardwaremonitor/openhardwaremonitor #1061: Z370 Aorus Gaming 5

mcktech, 2017-12-28:
I would love to see Z370 support, I have an Aorus Z370 gaming 5, and has a lot of sensors, but without OHM i feel lost!.. (came from Z77 + 3570k)

Thanks and great work for one of the best swiss army knife for advance users!

[OpenHardwareMonitor.Report.txt](https://github.com/openhardwaremonitor/openhardwaremonitor/files/1590627/OpenHardwareMonitor.Report.txt)

DarthDestroyer, 2017-12-28:
try the latest Version, atleast for your cpu http://openhardwaremonitor.org/files/openhardwaremonitor-v0.8.0.3-alpha.zip

A new Report would be cool!

mcktech, 2017-12-30:
Cool!..  0.8.0.3-alpha nails it!
Some stuff that i saw:

  +- Memoria Libre  :   2727.1  2726.32  2729.36 (/nvidiagpu/0/smalldata/1)
|  +- GPU Memory Used :  344.902  342.641  345.684 (/nvidiagpu/0/smalldata/2)
|  +- GPU Memory Total :     3072     3072     3072 (/nvidiagpu/0/smalldata/3)

_Values_ are not showing in gadget. 

************************************
and this is weird:

+- Disco D-F  (/hdd/0)
|  +- Temp           :       36       36       36 (/hdd/0/temperature/0)
|  +- Usado          :  46.4308  46.4307  46.4309 (/hdd/0/load/0)
|  +- Total Bytes Written :     9232  9231.98     9232 (/hdd/0/data/0)  **<=== THIS** 
|
+- Disco C SSD (/hdd/1)
|  +- Temp           :       34       34       34 (/hdd/1/temperature/0)
|  +- Usado          :  90.6965  90.6965  90.6965 (/hdd/1/load/0)

"Disco D-F" is a mechanical...  and "Disco C SSD" is a SSD, is the node misplaced?

Thanks again!

[OpenHardwareMonitor.Report.txt](https://github.com/openhardwaremonitor/openhardwaremonitor/files/1594384/OpenHardwareMonitor.Report.txt)

New report...

Sharakkhal, 2017-12-31:
I'm also missing OpenHardwareMonitor for monitoring temps and fans and such after upgrading to the Z370 Aorus Gaming 5 from a Sandy Bridge platform. Just in case another report may help, I'm attaching one. With the temperature probe connections, it would be nice to have those show up to monitor water temperature within OpenHardwareMonitor.

Also, it looks like the Samsung EVO 960 SSD isn't showing any monitoring info, just shows as a generic hard disk.

Let me know if you need more info, any help is appreciated.

Thanks!

[OpenHardwareMonitor.Report.Sharakkhal.txt](https://github.com/openhardwaremonitor/openhardwaremonitor/files/1595203/OpenHardwareMonitor.Report.Sharakkhal.txt)

DarthDestroyer, 2017-12-31:
Chip ID: Unknown ITE with ID 0x8686 at 0x2E/0x2F

Chip ID: Unknown Winbond / Nuvoton / Fintek with ID 0x8733 at 0x4E/0x4F

Chip ID: Unknown SMSC with ID 0x8733 at 0x4E/0x4F

mcktech, 2018-01-11:
Gather this from Aida .. if it helps
[Report_dmi.txt](https://github.com/openhardwaremonitor/openhardwaremonitor/files/1623619/Report_dmi.txt)
[Report_sensors.txt](https://github.com/openhardwaremonitor/openhardwaremonitor/files/1623620/Report_sensors.txt)

Sharakkhal, 2018-01-14:
I managed to install Visual Studio and download the source and compile it. Then I added in the IT8686E sensor using the instructions from #970 and adding the chip code and ignoring the CPU changes, so I get that information now. I also figured out that there's a 6th temperature from that chip for my board and made that array bigger to get it. However, it appears this board has a second chip, the IT8792E with the ID of 0x8733. I haven't found anything by searching on how I would get OHM to read the information from both chips. There's another set of voltages, temps, and fans which are read/controlled by that second chip.

Has anyone managed to get OHM to do that, and can someone point me in that direction?

Thanks!

Here's an updated report after adding the IT8686E chip support.
[OpenHardwareMonitor.Report3.txt](https://github.com/openhardwaremonitor/openhardwaremonitor/files/1629479/OpenHardwareMonitor.Report3.txt)

DarthDestroyer, 2018-01-14:
It seems to be a Fan Controller, there is already #900 for that

Sharakkhal, 2018-01-14:
I had indeed seen that from 2016, but I was hoping I was just overlooking the answer to get this working.

Thanks for your help.
[image]

mcktech, 2018-01-23:
I Manually added IT8686E support like Sharakkal did... Compiled and run perfect...  But also cant find how to add IT8792E.

And: DarthDestroyer, I dont know what to do with this info you posted:

Chip ID: Unknown ITE with ID 0x8686 at 0x2E/0x2F
Chip ID: Unknown Winbond / Nuvoton / Fintek with ID 0x8733 at 0x4E/0x4F
Chip ID: Unknown SMSC with ID 0x8733 at 0x4E/0x4F

Sorry for my ignorance, but is there a page that show how to use or analyze this info?... Thanks for all you time!

DarthDestroyer, 2018-01-23:
I extract those Lines out of the Report, that we can find it later through the GitHub Search!
This is my first Step, i do a quick search after that to find Issues with the same IDs to Link them together.
When i'm on mobile, i dont add something like "Please Ignore, only for Search" or similar ;)

BTT:
I dont find any new or other informations about the second Chip, sorry!
You can always try it out by yourself! Add the unknown ID like the other one and look, what happens.
I dont use a Board with such a Chip, that i could try it out myself...

mcktech, 2018-01-26:
OK.. i m working on that.. but i have problem.

in the function 
`private bool DetectIT87(LPCPort port)`

there's a:
```
if (port.RegisterPort != 0x2E)
		return false;
```
and IT8782 (aka 0x8733) looks like uses 4E, so, that function exits when is not 2E..  it's safe to add an exception to allow 4E/4F??

Thanks.!

mcktech, 2018-01-26:
I think OHM it's not quite ready to read more than one chip at the same time....