openhardwaremonitor/openhardwaremonitor #1064: OpenHardware Can't read Nvidia P106

AliminGuntoro, 2017-12-29:
Hello , could somebody explain why OpenHardware can't read P106 Nvidia.
Thanks

DarthDestroyer, 2017-12-30:
If you would provide a Report, maybe ;)

twilingway, 2018-01-01:
P104-100 need too

DarthDestroyer, 2018-01-01:
Guys... Without a Report its digging in the dark!

twilingway, 2018-01-01:
What do you need for the report? I can give it after the 6th of January

DarthDestroyer, 2018-01-01:
A Report is a Textfile created out of OHM!
There is more Data listed in, which maybe lead to the Malfunction!

ThePit12, 2018-01-02:
Hi!
some info about 106-100

[1066 report.txt](https://github.com/openhardwaremonitor/openhardwaremonitor/files/1597090/1066.report.txt)
[image]
[image]
[image]

DarthDestroyer, 2018-01-02:
GPU Data is available and OHM seems to get these Infos!

I dont think it will help with that, but you use an outdated Version

twilingway, 2018-01-02:
Where Can I download new dll? I have 0.8.0.0

DarthDestroyer, 2018-01-02:
http://openhardwaremonitor.org/files/openhardwaremonitor-v0.8.0.3-alpha.zip

ThePit12, 2018-01-02:
[1066_0803Areport.txt](https://github.com/openhardwaremonitor/openhardwaremonitor/files/1597496/1066_0803Areport.txt)

twilingway, 2018-01-02:
We dont see GPU in report and program

DarthDestroyer, 2018-01-02:
You do! Just scroll down in the Report ;-)

It seems that the Problem is OHM related, give some time

twilingway, 2018-01-02:
I scroll  but I see only one GPU

DarthDestroyer, 2018-01-02:
If there should be more, you could have said that!
Or do i missunderstand you?

twilingway, 2018-01-02:
We have 6 GPU one 1060 6g and 5 p106-100 in program and report we see only one 1060 nvidia

DarthDestroyer, 2018-01-02:
aye, i understood!

then, it isnt ohm's fault ;-)
it seems so, that the nvidia api only reports one gpu :-|

twilingway, 2018-01-02:
in gpu_z and hwinfo64 see this is gpu

DarthDestroyer, 2018-01-04:
they use not the nvapi, maybe?

As you can see:
NVAPI

Version: NVidia Complete Version 1.10
Number of GPUs: 1

--------------------------------------------------------------------------------

Nvidia GPU

Name: NVIDIA GeForce GTX 1060 6GB
Index: 0
Driver Version: 388.59
Driver Branch: r388_43-11

I dont think, that i had seen such a case... 
Could need some googling here, if NVAPI supports them or not!

ThePit12, 2018-01-06:
[1066_0803A_drv 388.71 report.txt](https://github.com/openhardwaremonitor/openhardwaremonitor/files/1608941/1066_0803A_drv.388.71.report.txt)
after driver update. no changes

DarthDestroyer, 2018-01-06:
Yeah... I think i know, where the problem is ;)

There was a similar Problem here => https://forum-en.msi.com/index.php?topic=292187.0

Well, the "GPU" Driver is working well, but only for the "GPU"!
In that Thread they say, that there isnt a "real Driver" existing for that Card?
Have you installed a seperate Driver for them?

It seems to be no wonder, that the NVAPI doesnt detect those :(

ThePit12, 2018-01-07:
I installed only manli, its manufacturer software, to increase mem and core speed. but may be driver is inside, theres a link https://yadi.sk/d/Knm5QOsW3R45Q3 password is 1

DarthDestroyer, 2018-01-07:
On some other Sites they said, that you can install any Driver you want!
It shouldnt depend on the manufacturer...

You could try that, atleast!

AliminGuntoro, 2018-01-09:
[image]

I try to make GUI program using c# with OpenHardwareMonitor, but it seem not show for NVIDiA p106, for GTX1070 is showing well.
This program i made to reset or shutdown or make powerON/OFF using Relay , you know sometimes mining software hank, i just want to reset or etc ,  and remote using my cellphone, that's why i need information load for each GPU card and send info to my cellphone, please see below what i mean

twilingway, 2018-01-25:
Has anyone solved the problem with the new drivers?

DarthDestroyer, 2018-01-25:
As long as no one would check that and try atleast some drivers, no!

How should anyone got a result without a mining card? i dont own one and i cant try it! 

Like i said, it is a known problem from some drivers (found in under 1 minute through google)! 

i would love to help you, but i am unable to test nothing ;-)

NVAPI doesnt work on that Cards <- a startpoint for google and/or card manufacturer!

AshotN, 2018-08-27:
NVAPI does not pick these cards up, but [NVML ](https://docs.nvidia.com/deploy/nvml-api/) does. So that would be the only way to add support, but quite some work since the DLL is for C++

GrapeLUX, 2019-03-13:
any updates on this? i have a p106 that can be used for testing

AshotN, 2019-03-14:
This is a very niche issue, I doubt anyone is willing to pick it up.

DarthDestroyer, 2019-03-14:
The Nvidia Support does need some love, that isnt wrong!
If that means support for the P106, maybe...

We will have to wait at this point!

MmAaXx500, 2019-03-14:
Take a look to Libre hardware monitor (fork of ohm)