openhardwaremonitor/openhardwaremonitor #1065: CPU info shows 4 cores only

SnecxXx, 2017-12-30:
Hello

I have a ASUS STRIX Z270F GAMING and a i7 7700k. With the latest OHM 0.8.1 i can see 4 cores only.
The mainboard have a lot more sensors but OHM shows no sensors.

Maybe someone can have a look into it. 

Attached the log.
[OpenHardwareMonitor.Report.txt](https://github.com/openhardwaremonitor/openhardwaremonitor/files/1594122/OpenHardwareMonitor.Report.txt)

DarthDestroyer, 2017-12-30:
1. You use a 4 Core CPU (OHM Displays only Real Cores!) => https://ark.intel.com/products/97129/Intel-Core-i7-7700K-Processor-8M-Cache-up-to-4_50-GHz

2. For Mainboard => #981

Chip ID: Unknown Winbond / Nuvoton / Fintek with ID 0xD121 at 0x2E/0x2F

SnecxXx, 2017-12-30:
Thank you. I didn't know OHM shows the real cores only and not all threads.
I have merged all PRs and compiled a version by myself. Now i have support for my mainboard and i can 
read some maniboard sensors.
[image]