openhardwaremonitor/openhardwaremonitor #1066: How to read powers value?

MHShafi, 2018-01-04:
In ohm, there is a field called Powers under CPU that gives energy consumption in watt. Now i want to use these values in my java application. But i did not find any interface to use this value in other application as per documentation section of the website. Please let me know how can i use this value? @DarthDestroyer can you please help me in this issue? Thanks in advance.

DarthDestroyer, 2018-01-04:
Could you explain a little bit more?

Some Points are missing! 
How do you retrieve the Data in Java?
Could you provide an code example? 
Which Documentation do you mean? 

Java isnt my Language, but of course i will try to help!

MHShafi, 2018-01-04:
I want the CPU package, cores, graphics and DRAM value which is already showing in ohm in watt.  Now I want to use this sensor value in other application. So for that reason i need to access ohm from my program. And in ohm documentation listed in the website, http://openhardwaremonitor.org/wordpress/wp-content/uploads/2011/04/OpenHardwareMonitor-WMI.pdf ; is mentioned i can access sensor data. But using WMI i can query other sensor information such as temperature, load but not powers. I want to get the sensor value of "Powers" field of ohm application. Hope now you get my point.

DarthDestroyer, 2018-01-04:
Yeah, i think i got you ;)
There are a lot of ways, to get the data... You choose the WMI Way - that will help!

As you can see, the PDF is from 2011. There are some new SensorTypes!

```
public enum SensorType {
    Voltage, // V
    Clock, // MHz
    Temperature, // °C
    Load, // %
    Fan, // RPM
    Flow, // L/h
    Control, // %
    Level, // %
    Factor, // 1
    Power, // W
    Data, // GB = 2^30 Bytes    
    SmallData, // MB = 2^20 Bytes 
  }
```

Did you use a WMI Explorer (or something) to get all available Items? Which selector do you use?

MHShafi, 2018-01-04:
Thanks for the reply and now i can query the power data. And yes, i am using WMI Explorer. What other ways are there to get the data please tell me. I am using WMI because i find it in the documentation. And for accessing WMI with JAVA is difficult. So i am really suffering a lot. But if you can refer me some other ways to get these data without using WMI it would be really great.  Thanks again.

DarthDestroyer, 2018-01-04:
First Option would be Java interoperation with a Native DLL using JNA.
Second Option parse the build in webpage.
Or you could write a vb/c# (or other) handler who handles the data and gives it over to you.

Many Ways to go! But afaik, the DLL Way would hurt a lot - i dont know, how deep you are into Java!?
The Pro about this way is, OHM does not have to run because you fetch the data through the lib.dll!

Could explain more, but im a little bit busy :/

MHShafi, 2018-01-04:
@DarthDestroyer  Thanks anyway

closed 2018-01-04 by MHShafi