openhardwaremonitor/openhardwaremonitor #1067: CPU temperature not reported on new Intel NUC NUC6CAYB

rayrayrayraydog, 2018-01-04:
I have just gotten a new NUC NUC6CAYB running with Windows 10 Pro 1709.  With OpenHardwareMonitor 0.8.0.3 Alpha (downloaded from comments here), I am not getting any CPU temperature reading.  I have submitted a report through the File->Submit Report feature within the app, but I can also post the report file here if necessary.

Thanks for the work on this program!

DarthDestroyer, 2018-01-04:
it would be nice, if you would post it here too! ;-)

rayrayrayraydog, 2018-01-04:
Here it is.  Thanks!
[NUC_NUC6CAYB_openhardwaremonitor.txt](https://github.com/openhardwaremonitor/openhardwaremonitor/files/1604916/NUC_NUC6CAYB_openhardwaremonitor.txt)

DarthDestroyer, 2018-01-04:
What Device is that?

Chip ID: Unknown Winbond / Nuvoton / Fintek with ID 0x8987 at 0x4E/0x4F

Chip ID: Unknown SMSC with ID 0x8987 at 0x4E/0x4F

rayrayrayraydog, 2018-01-05:
The same chip seems to have already come up in another issue: https://github.com/openhardwaremonitor/openhardwaremonitor/issues/960

It sounds like there's no support for it.  Oh well.

DarthDestroyer, 2018-01-25:
IT8987D a quick google said that and it seems to be an embedded controller...
As always, no datasheet