openhardwaremonitor/openhardwaremonitor #1068: Harddisk-Temperature not correct

Anacunga, 2018-01-05:
[image]
[OpenHardwareMonitor.Report.txt](https://github.com/openhardwaremonitor/openhardwaremonitor/files/1606340/OpenHardwareMonitor.Report.txt)

[image]

Btw.: would be nice if USB Harddisks would a) also set in the list and b) could also show thempearture ... yep, I know that the external HD has to support that ... ;)

DarthDestroyer, 2018-01-05:
that is weard because all other hdds are working fine! 

there are a lot of possible Problems! Controller, other running tools or the Mainboard / HDD itself...

Other Tools may retrieve the Data on other ways or do have a better error handling (alternative method).

i have to try it out, im using several hdds too!

Anacunga, 2018-01-07:
I did some changes in the HD configuration what did not change anything for quite a while ... but after going to sleep and waking up I made a look at it – and the temperature is back ... btw: there was no S.M.A.R.T.  issue (all HDs passed OK) ...