openhardwaremonitor/openhardwaremonitor #1069: ASUS TUF Z270 MARK 2 and Samsung SSD 960 EVO 500GB

ManiaProductions, 2018-01-05:
hello, can you please tell me or direct me to the solution, i can't see any sensors for my ssd, is it supported on said mobo? what about RAM? It just says Generic Memory and Generic Hard Disk...

special thanks to @DarthDestroyer for explanation on how to change the code to see the sensors on my mobo, in the #981 and #884  
i'm a complete noob with microsoft visual and coding, and i got it in like an hour, and the sensors are working for mobo! you're a real lifesaver! 
so if anyone has solution can you please direct me to it?
thx

DarthDestroyer, 2018-01-05:
Could you provide a Report?

Thanks for the nice Words and good job! ;-)

ManiaProductions, 2018-01-05:
[OpenHardwareMonitor.Report.txt](https://github.com/openhardwaremonitor/openhardwaremonitor/files/1606694/OpenHardwareMonitor.Report.txt)

i hope im doin this right :)

DarthDestroyer, 2018-01-05:
Did you compile from the latest Source-Code?
The Mainboard shouldnt be a Problem, in that case

Generic Memory = RAM
Generic Hard Disk = HDD where not more Data could be got from (for example, unsupported)

ManiaProductions, 2018-01-05:
not so sure anymore... i downloaded OHM from one of the issues submitted, it was an alpha version.
then in MV in OpenHardwareMonitorLib did all the changes for my mobo and build it.
then found the new Lib build and replaced the one in OHM folder from the alpha version.
it worked for the mobo so i thought that's the way you do it?
is that correct?
not getting info on Samsung SSD and my RAM...

ManiaProductions, 2018-01-05:
ok, sorry, I keep mentioning samsung ssd, my samsung device is in fact NVMe M.2, can IHM detect it as it is and show temperature, besides Load(i suppose the Generic Hard Disk Load is the M.2 Load)

DarthDestroyer, 2018-01-05:
That could make sense, because NVMe isnt implemented yet, but is already pulled here #984 

It is worth a try, i think!

ManiaProductions, 2018-01-06:
i tried #984 and unfortunately it doesn't work. No hard drive is detected and nothing is shown, not even Generic Hard Disk. here is the report.
[OpenHardwareMonitor.Report.txt](https://github.com/openhardwaremonitor/openhardwaremonitor/files/1608444/OpenHardwareMonitor.Report.txt)

i managed to patch up few other things so i think i followed on #984 as i should. but it's not working.
do you have any ideas what am i doing wrong? thanks for your help

ManiaProductions, 2018-01-06:
dude! i finaly got it. i'm the noobest of the noobs :) my nvme driver was missing :facepalm:

Thank you so much for your help man, you're the best!

one more question tho, is it possible to see the power consumption sensor for the GPU?
ie, in hwinfo i can see the GPU Power sensor, can i get it in OHM?
thanks again

DarthDestroyer, 2018-01-07:
nice to hear that, well done!

for the gpu data, i link it here also > #787

closed 2018-01-07 by ManiaProductions

DevilsJoker, 2019-12-19:
> dude! i finaly got it. i'm the noobest of the noobs :) my nvme driver was missing :facepalm:
>
> Thank you so much for your help man, you're the best!
>
> one more question tho, is it possible to see the [...]

I'm having the same issue with nothing showing up. What driver are you talking about?