openhardwaremonitor/openhardwaremonitor #1070: Gigabyte B250M-D3V-CF

FedyaLutkovski, 2018-01-10:
Hello!
On this motherboard, sensors are not displayed. In general, there are no

DarthDestroyer, 2018-01-10:
It would be nice, if you could upload a Report!

FedyaLutkovski, 2018-01-10:
Open Hardware Monitor Report

--------------------------------------------------------------------------------

Version: 0.8.0.3

--------------------------------------------------------------------------------

Common Language Runtime: 4.0.30319.42000
Operating System: Microsoft Windows NT 10.0.16299.0
Process Type: 64-Bit

--------------------------------------------------------------------------------

Sensors

|
+- Gigabyte B250M-D3V-CF (/mainboard)
|
+- Intel Celeron G3900 (/intelcpu/0)
|  +- Bus Speed      :  100.286  100.286  100.288 (/intelcpu/0/clock/0)
|  +- CPU Core #1    :     2808  802.301     2808 (/intelcpu/0/clock/1)
|  +- CPU Core #2    :     2808  802.301     2808 (/intelcpu/0/clock/2)
|  +- CPU Core #1    :       26       25       31 (/intelcpu/0/temperature/0)
|  +- CPU Core #2    :       25       25       28 (/intelcpu/0/temperature/1)
|  +- CPU Package    :       28       27       31 (/intelcpu/0/temperature/2)
|  +- CPU Total      :   7.8125   1.5625       60 (/intelcpu/0/load/0)
|  +- CPU Core #1    :    3.125        0      100 (/intelcpu/0/load/1)
|  +- CPU Core #2    :     12.5        0  66.6667 (/intelcpu/0/load/2)
|  +- CPU Package    :  2.16107  1.61448  7.33118 (/intelcpu/0/power/0)
|  +- CPU Cores      :   1.2796 0.770877  5.95524 (/intelcpu/0/power/1)
|  +- CPU Graphics   : 0.0776477        0 0.498264 (/intelcpu/0/power/2)
|  +- CPU DRAM       : 0.336351 0.281589 0.888946 (/intelcpu/0/power/3)
|
+- Generic Memory (/ram)
|  +- Memory         :  43.6445  43.6445  43.6741 (/ram/load/0)
|  +- Used Memory    :  1.69807  1.69807  1.69922 (/ram/data/0)
|  +- Available Memory :  2.19261  2.19146  2.19261 (/ram/data/1)
|
+- Samsung SSD 850 EVO 250GB (/hdd/0)
|  +- Temperature    :       27       27       27 (/hdd/0/temperature/0)
|  +- Used Space     :  13.6593  13.6593  13.6593 (/hdd/0/load/0)
|  +- Total Bytes Written :  70.1772  70.1772  70.1772 (/hdd/0/data/0)

--------------------------------------------------------------------------------

Parameters

|
+- Gigabyte B250M-D3V-CF (/mainboard)
|
+- Intel Celeron G3900 (/intelcpu/0)
|  |
|  +- CPU Core #1 (/intelcpu/0/temperature/0)
|  |  +- TjMax [°C] : 100 : 100
|  |  +- TSlope [°C] : 1 : 1
|  |
|  +- CPU Core #2 (/intelcpu/0/temperature/1)
|  |  +- TjMax [°C] : 100 : 100
|  |  +- TSlope [°C] : 1 : 1
|  |
|  +- CPU Package (/intelcpu/0/temperature/2)
|  |  +- TjMax [°C] : 100 : 100
|  |  +- TSlope [°C] : 1 : 1
|
+- Generic Memory (/ram)
|
+- Samsung SSD 850 EVO 250GB (/hdd/0)
|  |
|  +- Temperature (/hdd/0/temperature/0)
[... 290 lines cut ...]
SSDSamsung

Drive name: Samsung SSD 850 EVO 250GB
Firmware version: EMT03B6Q

 ID Description                        Raw Value    Worst Value Thres Physical
 05 Reallocated Sectors Count          000000000000 100   100   10    -       
 09 Power-On Hours (POH)               650000000000 99    99    0     101     
 0C Power Cycle Count                  080000000000 99    99    0     8       
 B1 Wear Leveling Count                000000000000 100   100   0     0       
 B3 Used Reserved Block Count (Total)  000000000000 100   100   10    0       
 B5 Program Fail Count (Total)         000000000000 100   100   10    0       
 B6 Erase Fail Count (Total)           000000000000 100   100   10    0       
 B7 Runtime Bad Block Total            000000000000 100   100   10    0       
 BB Uncorrectable Error Count          000000000000 100   100   0     0       
 BE Temperature                        1B0000000000 63    73    0     27      
 C3 ECC Rate                           000000000000 200   200   0     -       
 C7 CRC Error Count                    000000000000 100   100   0     0       
 EB Power Recovery Count               030000000000 99    99    0     -       
 F1 Total LBAs Written                 35ADC5080000 99    99    0     70.17739

Logical drive name: C:\
Format: NTFS
Total size: 249531723776
Total free space: 215447560192

DarthDestroyer, 2018-01-10:
Chip ID: Unknown ITE with ID 0x8686 at 0x2E/0x2F

#970 will help you!
There is also an instruction on how you could do that, in that PR!

DarthDestroyer, 2018-01-11:
Does it work? If yes, nice to hear that ;-)

FedyaLutkovski, 2018-01-11:
Yes, it work =)