openhardwaremonitor/openhardwaremonitor #1071: No support for i9 7900x Temps

ProtostarMusic, 2018-01-11:
Only core loads show, no temps/voltages etc.
Shows fine in HWMonitor etc!

https://i.imgur.com/ozJVAHS.png

DarthDestroyer, 2018-01-11:
Without a Report, there is only one Option :-P

Do you use the latest Version?
http://openhardwaremonitor.org/files/openhardwaremonitor-v0.8.0.3-alpha.zip

ProtostarMusic, 2018-01-11:
Yup!
https://i.imgur.com/S2qX7Jh.png

Here's a report: https://pastebin.com/LTEW0e8H

ProtostarMusic, 2018-01-18:
@DarthDestroyer any help on the matter? :)

DarthDestroyer, 2018-01-18:
I can only lead you to #1005, it should help you with the CPU Data

ProtostarMusic, 2018-01-18:
@DarthDestroyer  yeah this should help, i have no idea how to utilise this though sadly haha, i dont touch much to do with code, any tuts/advice? :D

DarthDestroyer, 2018-01-18:
in #981 is it a little bit explained, its not really complicated.

if you can write a word document, you would handle it ;-)

ProtostarMusic, 2018-01-18:
Any specific packs i need when installing VS? :D https://i.imgur.com/AOE58v8.png

DarthDestroyer, 2018-01-18:
I cant really read it because of the Resolution, but i think that "top right" is enough!

Should be .net or similar, if i remember correct

ProtostarMusic, 2018-01-18:
@DarthDestroyer  you can click on the image on imgur to enlarge it!

DarthDestroyer, 2018-01-18:
Not on mobile :-D

But, i think im correct

ProtostarMusic, 2018-01-18:
I think i understand how to change the lines in the code to correct the bug, but am i opening the executable up in visual studio? or is there a DLL i need to find

DarthDestroyer, 2018-01-18:
The fast and easiest way is:

Click on Run, wait for the compilation and look if it works. 
After that, go to the folder of the Sourcecode, there should be a new Folder... 

In that Folder should be a "OpenhardwaremonitorLib.dll" which you can copy and paste into your downloaded Folder and overwrite the "incomplete old" Lib ;-)

You could aswell use OHM from the Sourcecode Folder, too... It depends on you!

ProtostarMusic, 2018-01-18:
I seem to get an error of 'the version of the openhardwaremonitorlib.dll is incompatible. Im running the latest alpha exe too

ProtostarMusic, 2018-01-18:
Got it! I compiled the EXE and replaced everything there and its all working now, shows clock speeds temps and loads, thanks for the help!

DarthDestroyer, 2018-01-18:
awesome! 

welcome in the coding world ;-)

ProtostarMusic, 2018-01-18:
I did a year long module in the middle of my mechanical engineering degree on C++ haha! but its been several years since ive touched anything in that world so was extremely rusty haha.

DarthDestroyer, 2018-01-18:
then, let me correct myself... welcome back ;-)