openhardwaremonitor/openhardwaremonitor #1073: Gigabyte Z370P D3-CF not supported

Animoral, 2018-01-15:
Hello, 
There is no data for **Gigabyte Z370P D3-CF** motherboard (for intel gen 8). 
I'm using Version: 0.8.0.5.

A report is attached.
[OpenHardwareMonitor.Report.txt](https://github.com/openhardwaremonitor/openhardwaremonitor/files/1632066/OpenHardwareMonitor.Report.txt)

DarthDestroyer, 2018-01-15:
Chip ID: Unknown ITE with ID 0x8686 at 0x2E/0x2F

DarthDestroyer, 2018-01-15:
#970 will help you!
There is also an instruction on how you could do that, in that PR!

Animoral, 2018-01-15:
I'm sorry, what is PR? I'm new at github so I'm not familiar with what I should do.
I used the latest alpha version I'v e found (0.8.0.5).
Do you mean that I should somehow add Z370P D3-CF manually to the code and then compile it? If so, I don't really know how to do so...

DarthDestroyer, 2018-01-15:
A PR is a Pull Request (Something like "Code to Review")
It is everything done and you only have to add these Codelines to your local Source-Code!

In that PR is it explained, how you do that and what you need!
If you would trust an unknown Guy at the Internet, you could Request the lib dll from #1070

Animoral, 2018-01-15:
ok, sorry for the dumb questions...
Can I expect an Alpha version with Gigabyte Z370P D3-CF support?
If no, how can I request the lib dll from #1070 ? I don't see any downloadable files there, only someone's report and you referring him to #970 .
Thanks.

DarthDestroyer, 2018-01-15:
That guy could be contactable! Click on his Icon ;)

You could expect a Release, but there is no ETA