openhardwaremonitor/openhardwaremonitor #1078: 8700k not fully supported

DreddlockRasta, 2018-02-03:
The Intel Core i7 8700k CPU is not fully supported. The program recognises the 6 cores but does not show any threads at all.
The program also does not show any temperature reading for the 8700k.
Many thanks for taking the time to fix this

DarthDestroyer, 2018-02-03:
OHM Displays only Real Cores!
There is a Workaround available here: #954

A Report would be nice, if you miss more! ;-)

DreddlockRasta, 2018-02-03:
@DarthDestroyer A. How do go about implementing that fix because I dont even have the files it says I should have, any help would be greatly appreciated.
2. How do I go about submitting a report?

DarthDestroyer, 2018-02-03:
Everyone can get the Files, it is an OpenSource Tool! You can download the Sourcecode and play with it. It is a free Version of Visual Studio available from Microsoft itself.

More is not needed! ;-)

To your Second Part, click on Files > Report and save that txt-File within OHM. Drag & Drop that File here and we can look for your Issues! 

Regards

DreddlockRasta, 2018-02-03:
You sir are a legend, thank you so much for the information, I look forward to playing with the source code and if I get good results I shall post anything relevant here. 
I shall also upload the report and close this issue as soon as I have.
Many thanks again bud

DreddlockRasta, 2018-02-03:
@DarthDestroyer I have the Microsoft Visual Studio and have edited the source coding to allow for the changes but I am stumped as to how I can compile the source code into an executable program. Any advice bud?

DarthDestroyer, 2018-02-04:
That isnt really hard! Press the green Play Button and it will start.

After that, you will find a Bin Folder in the Sourcecode Directory, where the Files are.