openhardwaremonitor/openhardwaremonitor #1080: Webserver returning JS error: ": viewModel is not defined"

seanvree, 2018-02-13:
OS: Windows 10E 1709 FCU

ISSUE: webserver not populating data. 

Had this working for a year or two, not sure why this is happening now.

In addition, when I click on options -> remote web server -> PORT -> the app crashes with the error below:

```
Version: 0.8.0.3

System.ArgumentOutOfRangeException: Value of '8050' is not valid for 'Value'. 'Value' should be between 'Minimum' and 'Maximum'.
Parameter name: Value
   at System.Windows.Forms.NumericUpDown.set_Value(Decimal value)
   at OpenHardwareMonitor.GUI.PortForm.PortForm_Load(Object sender, EventArgs e)
   at System.Windows.Forms.Form.OnLoad(EventArgs e)
   at System.Windows.Forms.Form.OnCreateControl()
   at System.Windows.Forms.Control.CreateControl(Boolean fIgnoreVisible)
   at System.Windows.Forms.Control.CreateControl()
   at System.Windows.Forms.Control.WmShowWindow(Message& m)
   at System.Windows.Forms.Control.WndProc(Message& m)
   at System.Windows.Forms.Form.WmShowWindow(Message& m)
   at System.Windows.Forms.NativeWindow.Callback(IntPtr hWnd, Int32 msg, IntPtr wparam, IntPtr lparam)

```

However, when the app is first launched, the webserver is running on port 8065, but not serving data, the browser console returns these errors:

```
ohm_web.js:103 Uncaught ReferenceError: viewModel is not defined
    at HTMLDivElement.slide (http://localhost:8050/js/ohm_web.js:103:7)
    at b.(anonymous function).(anonymous function)._trigger (http://localhost:8050/js/jquery-ui-1.8.16.custom.min.js:33:438)
    at b.(anonymous function).(anonymous function)._slide (http://localhost:8050/js/jquery-ui-1.8.16.custom.min.js:102:453)
    at b.(anonymous function).(anonymous function)._mouseDrag (http://localhost:8050/js/jquery-ui-1.8.16.custom.min.js:100:54)
    at b.(anonymous function).(anonymous function)._mouseMove (http://localhost:8050/js/jquery-ui-1.8.16.custom.min.js:49:92)
    at HTMLDocument._mouseMoveDelegate (http://localhost:8050/js/jquery-ui-1.8.16.custom.min.js:48:252)
    at HTMLDocument.dispatch (http://localhost:8050/js/jquery-1.7.2.min.js:3:4816)
    at HTMLDocument.i (http://localhost:8050/js/jquery-1.7.2.min.js:3:709)

```

[image]

DarthDestroyer, 2018-02-16:
So, i tested it.

* I can change the Port to whatever Integer i want (btw: after Port change, restart the Webserver)
* I can run the webserver as it should

I think, that there is an error in the returning data!
Could you Post the "Response" Content from "/data.json" here?
(You posted Screenshots from the Dev-Tools, so i think you could find it?!)

Maybe, the fault is missing Hardware Support... A Report of your Configuration could help us, too!

seanvree, 2018-02-18:
@DarthDestroyer  

here's the response content:
```
{"id": 0, "Text": "Sensor", "Children": [{"id": 1, "Text": "VREEPLEXDESKTOP", "Children": [{"id": 2, "Text": "Gigabyte H170N-WIFI-CF", "Children": [{"id": 3, "Text": "ITE IT8628E", "Children": [{"id": 4, "Text": "Voltages", "Children": [{"id": 5, "Text": "Voltage #1", "Children": [], "Min": "0.084 V", "Value": "0.924 V", "Max": "1.320 V", "ImageURL": "images/transparent.png"}, {"id": 6, "Text": "Voltage #2", "Children": [], "Min": "2.016 V", "Value": "2.028 V", "Max": "2.028 V", "ImageURL": "images/transparent.png"}, {"id": 7, "Text": "Voltage #3", "Children": [], "Min": "2.004 V", "Value": "2.016 V", "Max": "2.016 V", "ImageURL": "images/transparent.png"}, {"id": 8, "Text": "Voltage #4", "Children": [], "Min": "2.004 V", "Value": "2.016 V", "Max": "2.016 V", "ImageURL": "images/transparent.png"}, {"id": 9, "Text": "Voltage #5", "Children": [], "Min": "0.012 V", "Value": "0.012 V", "Max": "1.152 V", "ImageURL": "images/transparent.png"}, {"id": 10, "Text": "Voltage #6", "Children": [], "Min": "1.080 V", "Value": "1.080 V", "Max": "1.080 V", "ImageURL": "images/transparent.png"}, {"id": 11, "Text": "Voltage #7", "Children": [], "Min": "1.200 V", "Value": "1.200 V", "Max": "1.200 V", "ImageURL": "images/transparent.png"}, {"id": 12, "Text": "Standby +3.3V", "Children": [], "Min": "3.384 V", "Value": "3.384 V", "Max": "3.384 V", "ImageURL": "images/transparent.png"}, {"id": 13, "Text": "VBat", "Children": [], "Min": "2.808 V", "Value": "2.808 V", "Max": "2.808 V", "ImageURL": "images/transparent.png"}], "Min": "", "Value": "", "Max": "", "ImageURL": "images_icon/voltage.png"}, {"id": 14, "Text": "Temperatures", "Children": [{"id": 15, "Text": "Temperature #1", "Children": [], "Min": "87.8 °F", "Value": "93.2 °F", "Max": "104.0 °F", "ImageURL": "images/transparent.png"}, {"id": 16, "Text": "MB Temp", "Children": [], "Min": "102.2 °F", "Value": "107.6 °F", "Max": "118.4 °F", "ImageURL": "images/transparent.png"}], "Min": "", "Value": "", "Max": "", "ImageURL": "images_icon/temperature.png"}, {"id": 17, "Text": "Fans", "Children": [{"id": 18, "Text": "Fan #1", "Children": [], "Min": "1395 RPM", "Value": "1421 RPM", "Max": "1452 RPM", "ImageURL": "images/transparent.png"}, {"id": 19, "Text": "Fan #2", "Children": [], "Min": "1790 RPM", "Value": "1819 RPM", "Max": "1844 RPM", "ImageURL": "images/transparent.png"}], "Min": "", "Value": "", "Max": "", "ImageURL": "images_icon/fan.png"}, {"id": 20, "Text": "Controls", "Children": [{"id": 21, "Text": "Fan Control #1", "Children": [], "Min": "-", "Value": "-", "Max": "-", "ImageURL": "images/transparent.png"}, {"id": 22, "Text": "Fan Control #2", "Children": [], "Min": "-", "Value": "-", "Max": "-", "ImageURL": "images/transparent.png"}, {"id": 23, "Text": "Fan Control #3", "Children": [], "Min": "-", "Value": "-", "Max": "-", "ImageURL": "images/transparent.png"}], "Min": "", "Value": "", "Max": "", "ImageURL": "images_icon/control.png"}], "Min": "", "Value": "", "Max": "", "ImageURL": "images_icon/chip.png"}], "Min": "", "Value": "", "Max": "", "ImageURL": "images_icon/mainboard.png"}, {"id": 24, "Text": "Intel Core i7-6700K", "Children": [{"id": 25, "Text": "Clocks", "Children": [{"id": 26, "Text": "Bus Speed", "Children": [], "Min": "100 MHz", "Value": "100 MHz", "Max": "100 MHz", "ImageURL": "images/transparent.png"}, {"id": 27, "Text": "CPU Core #1", "Children": [], "Min": "802 MHz", "Value": "4008 MHz", "Max": "4208 MHz", "ImageURL": "images/transparent.png"}, {"id": 28, "Text": "CPU Core #2", "Children": [], "Min": "802 MHz", "Value": "4008 MHz", "Max": "4208 MHz", "ImageURL": "images/transparent.png"}, {"id": 29, "Text": "CPU Core #3", "Children": [], "Min": "802 MHz", "Value": "4008 MHz", "Max": "4208 MHz", "ImageURL": "images/transparent.png"}, {"id": 30, "Text": "CPU Core #4", "Children": [], "Min": "802 MHz", "Value": "4008 MHz", "Max": "4208 MHz", "ImageURL": "images/transparent.png"}], "Min": "", "Value": "", "Max": "", "ImageURL": "images_icon/clock.png"}, {"id": 31, "Text": "Temperatures", "Children": [{"id": 32, "Text": "CPU Core #1", "Children": [], "Min": "66.2 °F", "Value": "77.0 °F", "Max": "149.0 °F", "ImageURL": "images/transparent.png"}, {"id": 33, "Text": "CPU Core #2", "Children": [], "Min": "66.2 °F", "Value": "78.8 °F", "Max": "150.8 °F", "ImageURL": "images/transparent.png"}, {"id": 34, "Text": "CPU Core #3", "Children": [], "Min": "66.2 °F", "Value": "75.2 °F", "Max": "136.4 °F", "ImageURL": "images/transparent.png"}, {"id": 35, "Text": "CPU Core #4", "Children": [], "Min": "64.4 °F", "Value": "73.4 °F", "Max": "140.0 °F", "ImageURL": "images/transparent.png"}, {"id": 36, "Text": "CPU Package", "Children": [], "Min": "75.2 °F", "Value": "84.2 °F", "Max": "150.8 °F", "ImageURL": "images/transparent.png"}], "Min": "", "Value": "", "Max": "", "ImageURL": "images_icon/temperature.png"}, {"id": 37, "Text": "Load", "Children": [{"id": 38, "Text": "CPU Total", "Children": [], "Min": "2.3 %", "Value": "20.1 %", "Max": "96.6 %", "ImageURL": "images/transparent.png"}, {"id": 39, "Text": "CPU Core #1", "Children": [], "Min": "3.1 %", "Value": "21.4 %", "Max": "100.0 %", "ImageURL": "images/transparent.png"}, {"id": 40, "Text": "CPU Core #2", "Children": [], "Min": "1.4 %", "Value": "19.9 %", "Max": "97.3 %", "ImageURL": "images/transparent.png"}, {"id": 41, "Text": "CPU Core #3", "Children": [], "Min": "1.5 %", "Value": "17.7 %", "Max": "95.5 %", "ImageURL": "images/transparent.png"}, {"id [... cut ...] x": "94.5 %", "ImageURL": "images/transparent.png"}], "Min": "", "Value": "", "Max": "", "ImageURL": "images_icon/load.png"}], "Min": "", "Value": "", "Max": "", "ImageURL": "images_icon/hdd.png"}], "Min": "", "Value": "", "Max": "", "ImageURL": "images_icon/computer.png"}], "Min": "Min", "Value": "Value", "Max": "Max", "ImageURL": ""}
```

I wasn't sure how to create a "configuration report", but here is the output "save report"

https://gist.github.com/seanvree/13f92db466bef85514a85509c2510993

DarthDestroyer, 2018-02-18:
Could you try to rename GenericHarddisk?

Drive name: mÑÂÂõÿÿ

seanvree, 2018-02-18:
@DarthDestroyer  yup, that did it. Thanks.

closed 2018-02-18 by seanvree