openhardwaremonitor/openhardwaremonitor #1085: Implementation problem

110mat110, 2018-02-25:
I am trying to use library in my own aplication. I want to read CPU temp (and then send it to arduino) 

In app, I can see CPU temp 
[image]

but in my programmed app, I do not see nor temps nor clock
[image]

Code is simple.
https://paste.ofcode.org/jauWtisYf7mQaBkKE2jfxG

rmattuschka, 2018-02-25:
Your App must have admin-rights to read this values too.

110mat110, 2018-02-25:
Admin-rights are not enough. I used 
`<requestedExecutionLevel level="highestAvailable" uiAccess="false" /> `

Now I can read everything I wanted
[image]

closed 2018-02-25 by 110mat110