openhardwaremonitor/openhardwaremonitor #1088: GPU List

twilingway, 2018-03-01:
Hello.
By what principle is the list of the GPU formed?
If you open the MSI Afferberner or GPU_Z, then the GPU list is 0 1 2 3 in and so on. And then it's like a mess. Help to form a list of the same as in these programs.

DarthDestroyer, 2018-03-01:
I dont know "what you use" or where "this" come from, but the Original Lib is not named "openhwm.dll"!

But more to your Problem, what exactly do you mean?
OHM starts Counting at 0 - like the other tools, too!

What do you see? Could you provide a Screenshot or atleast a Report here?

twilingway, 2018-03-01:
https://yadi.sk/i/a4--LErz3Sv9VJ

DarthDestroyer, 2018-03-01:
OK, i try to understand you... You say, that the List of GPUs is reversed and you want it "the other way"?

If you would Post a Report (or read it), you will see a NVidia Part at the Bottom. There is a List, what the Driver gives OHM!

AMD Cards are a little bit wrong, atm! Had seen some Problems with them already!

twilingway, 2018-03-01:
Nvidia works well but the report starts from the bottom up and this is not a problem.
but AMD shows an incomprehensible

twilingway, 2018-03-02:
AB and gpu-Z put cards on the numbers of tires pci-E. But OHM hike from the dispatcher.

DarthDestroyer, 2018-03-02:
im currently working on the nvidia vcore sensor... as for nvidia it seems to be very outdated!

but if you could provide us reports from these Systems, it could help at this Point

twilingway, 2018-03-02:
How Can I get right reports?

DarthDestroyer, 2018-03-02:
In OHM => File => Save Report
Drag&Drop it here!

twilingway, 2018-03-02:
[OpenHardwareMonitor.Report.txt](https://github.com/openhardwaremonitor/openhardwaremonitor/files/1776064/OpenHardwareMonitor.Report.txt)

DarthDestroyer, 2018-03-02:
Had you tried the latest Version, too?

http://openhardwaremonitor.org/files/openhardwaremonitor-v0.8.0.3-alpha.zip

twilingway, 2018-03-02:
nothing changed
https://yadi.sk/i/Nljpwg743SxD58

[OpenHardwareMonitor.Report.txt](https://github.com/openhardwaremonitor/openhardwaremonitor/files/1776315/OpenHardwareMonitor.Report.txt)

twilingway, 2018-03-15:
Hi All.
Do you have news about it?

DarthDestroyer, 2018-03-15:
Im still working at nvidia atm because of some deprecated functions :-\

Sorry

emc02, 2018-03-27:
Numbering GPUs is really bad, especially AMD. I've got two cards 0 and 16?!
NVIDIA normally is ok with 0 and 1.

DarthDestroyer, 2018-03-27:
You could read your Report at the bottom... OHM can only handle what the driver/lib serves =(

After all that, OHM could ignore the other "gpus"... but it isnt really hard to work with the ghost gpus (i. e. index/16)...

This issue isnt handling this, it is about the indexing/numbering itself...

DarthDestroyer, 2018-04-05:
@twilingway

I had seen a similar "problem" at reddit! It is not really related to cuda our issue but worth a try to set the global var to try it out. 

https://www.reddit.com/r/gpumining/comments/8700fr/reorder_cuda_devices_to_match_device_manager/

AshotN, 2018-06-23:
I'm also exhibiting a similar problem where OHM lists my GPUs in a different order from every other software

My issue is with RX580s

twilingway, 2018-10-01:
Where I Can download last version OpenHardwareMonitorLib.DLL?

DarthDestroyer, 2018-10-01:
http://openhardwaremonitor.org/files/openhardwaremonitor-v0.8.0.3-alpha.zip is the latest compiled Version!
If you need something else, i recommend to do it by yourself! 
I would never load a library from an untrusted source!

twilingway, 2018-10-01:
I have this
http://openhardwaremonitor.org/files/openhardwaremonitor-v0.8.0.5-alpha.zip

DarthDestroyer, 2018-10-01:
i can, like you can do!

it is not possible for me to upload to the official webspace, sorry! 

.5 is even better, but not that big difference