openhardwaremonitor/openhardwaremonitor #1091: Core i9-7980XE and M/B Gigabyte X299 DESIGNARE EX-CF

dualazmak, 2018-03-19:
Core i9-7980XE and M/B Gigabyte X299 DESIGNARE EX-CF

I am using OHM  Ver. 0.8.0.Beta, and recently built new Workstation with CPU Intel Core i9-7980XE and M/B Gigabyte X299 DESIGNARE EX-CF.  The OHM Ver. 0.8.0 Beta only detect CPU cores 1 to 18 usage, but no other CPU settings nor any of the M/B sensors.  Do you have plan to update the distribution version of OHM soon to cover Core i9 series and X299 M/Bs?

dualazmak, 2018-03-19:
Here, I am attaching the OHM Report text on my Workstation;

[OpenHardwareMonitor.Report.txt](https://github.com/openhardwaremonitor/openhardwaremonitor/files/1825977/OpenHardwareMonitor.Report.txt)

DarthDestroyer, 2018-03-19:
There is one Option at that Point!

Did you use the latest Version? 0.8.0 isnt enough Info ;-)
http://openhardwaremonitor.org/files/openhardwaremonitor-v0.8.0.3-alpha.zip

For anything else we could need a Report ;-)

dualazmak, 2018-03-19:
OK, I installed OHM Ver. 0.8.0.3, and found the problems are still not solved.
I am attaching the OHM Report text by OHM 0.80.0.3

The M.2 SSD, MVMe Samsung SSD 960 (Drive C: with windows 10 Pro 64 bit OS installed) is also not fully detected, but detected as just "Generic Hard Disco (/hdd/2) with only the Used Space. 

[OpenHardwareMonitor.Report2.txt](https://github.com/openhardwaremonitor/openhardwaremonitor/files/1826008/OpenHardwareMonitor.Report2.txt)

dualazmak, 2018-03-22:
CPUID's free HWMonitor shows all the sensor values from M/B, CPU, M.2 SSD, HDD, and others, but HWMonitor has no Gadget display.  

I tentatively decided, however, to use CPUID's HWMonitor until OHM will be fully updated to next version with full major version-up.

rmattuschka, 2018-03-23:
Do not give up hope!

dualazmak, 2018-03-23:
Attached below are screen shots (into 3 parts) of CPUID's HWMonitor.  Some of the CPU and Thread nobering are odd/strange, but all the sensors are perfectly monitors, I believe.

[image]
[image]
[image]

also attached is CPUID's HWMonitor log text;
[HWMonitor_log.txt](https://github.com/openhardwaremonitor/openhardwaremonitor/files/1842024/HWMonitor_log.txt)