openhardwaremonitor/openhardwaremonitor #1094: CPU item numbering not consistent

emc02, 2018-03-27:
/intelcpu/0/load/0 => CPU Total
/intelcpu/0/load/1 => Core 1
/intelcpu/0/load/2 => Core 2
/intelcpu/0/load/3 => Core 3
/intelcpu/0/load/4 => Core 4
/intelcpu/0/load/5 => Core 5
/intelcpu/0/load/6 => Core 6

/intelcpu/0/temperature/0 => Core 1
/intelcpu/0/temperature/1 => Core 2
/intelcpu/0/temperature/2 => Core 3
/intelcpu/0/temperature/3 => Core 4
/intelcpu/0/temperature/4 => Core 5
/intelcpu/0/temperature/5 => Core 6
/intelcpu/0/temperature/6 => Package

wouldn't it be better to always make Index 1 => Core 1 and Index 0 is for the Package or CPU Total?!

DarthDestroyer, 2018-03-27:
the numbering isnt corresponding to another category...

for your example, Package != CPU Total.

and, as an example for maybe further implementations it would be a big pain, if there would be another temp sensor added! Where do you would put them? =/

For now, you should work with the string of the sensor aswell...

emc02, 2018-03-27:
on Clock speed there is also the BUS Speed on index 0 and then are the cores.
Maybe it would be good to make everything like the temperatures, so that the n cores are from 0 to n-1 and then everything else?
But it's ok, it was just an idea, because it's not consistent for automated data readout through wmic command.
The program itself is perfect and it's a great tool for me to read out all PCs!