openhardwaremonitor/openhardwaremonitor #1095: i7-8700k temps?

FoxiesFury, 2018-03-28:
I was surprised to see, actually "not see" tracking for this CPU, is it not supported yet?

FoxiesFury, 2018-03-28:
[OpenHardwareMonitor.Report.txt](https://github.com/openhardwaremonitor/openhardwaremonitor/files/1857639/OpenHardwareMonitor.Report.txt)

rmattuschka, 2018-03-31:
Are you kidding me?

FoxiesFury, 2018-03-31:
[image]

DarthDestroyer, 2018-03-31:
atleast your version is outdated > http://openhardwaremonitor.org/files/openhardwaremonitor-v0.8.0.3-alpha.zip

And if that doesnt help, take a look into the other Issues or Pull Requests ;-)

FoxiesFury, 2018-03-31:
It works now! Thanks. Why is there no download link for this version on site? Just because it's alpha build or?

DarthDestroyer, 2018-03-31:
it was a good try to get guys reading github, but instead everyone is posting new issues 😜

naaah, there is no real reason 😉

ziakhan110, 2018-06-27:
i had the same isssue with 7700H , v0.8.0.3-alpha fixed the issue.

greyhammer, 2019-02-05:
Same with i7-8750H.  v0.8.0.3-alpha shows the temps now.