openhardwaremonitor/openhardwaremonitor #1097: Nuvoton NCT6771F High temperatures

Stealhelm, 2018-04-07:
Getting temperatures on the #2 sensor at 110 - 125°C. The other two are at a cool ~40°-ish. CPU is at 42°C with dips into the mid 30s. Is this common or am I experiencing an bug?

System Information:
CPU - Intel I5 750, 4.0ghz
Motherboard - AsRock P55 Deluxe3
OS - 64-bit Windows 10 Pro, 10.0.16299, Build 16299

PS: Do apologize if this isn't an issue just seems really strange - I hope you can understand my concerns.

DarthDestroyer, 2018-04-07:
if you could upload a report, it could be looked at! ;)

Stealhelm, 2018-04-07:
> if you could upload a report, it could be looked at! ;)

Clocked down the CPU with no luck.
[OpenHardwareMonitor.Report.txt](https://github.com/openhardwaremonitor/openhardwaremonitor/files/1886512/OpenHardwareMonitor.Report.txt)

Stealhelm, 2018-04-07:
> atleast your version is outdated > http://openhardwaremonitor.org/files/openhardwaremonitor-v0.8.0.3-alpha.zip

Oh, sorry, I thought I downloaded the latest version.

> as i remember correct, there was a similar issue already?! nevermind! Could you try the latest Version?

Sure thing - here's the report:
[OpenHardwareMonitor.Report.txt](https://github.com/openhardwaremonitor/openhardwaremonitor/files/1886561/OpenHardwareMonitor.Report.txt)

DarthDestroyer, 2018-04-07:
The MoBo Sensors are sometimes a little bit weird... As long as your CPU Temp itself doesnt do crazy things, it isnt dangerous!

How do you know, that Temp#2 is a CPU related Sensor?

Stealhelm, 2018-04-07:
> How do you know, that Temp#2 is a CPU related Sensor?

Oh, I don't know that. I just assumed that seeing as the only thing I've changed was my OC it could've been interesting to know. Oh well. If it isn't dangerous I wont bother.

Thanks for the information - greatly appriciated! :)

pimentelleo, 2021-02-12:
I am facing the same, just downloaded the latest version. 
I was playing Astroneer, then the things goina crazy about 117°C

kayhen, 2021-09-16:
there are three NCT6779D with high temperature
79C and 93C and one of them is at 22C
But I don't know if this temperature is normal.

borovinskiy, 2021-11-10:
For my NCT6779D temp also incorrect:
CPU6: -23 C.
CPU3,4: 90+ C
[OpenHardwareMonitor.Report.txt](https://github.com/openhardwaremonitor/openhardwaremonitor/files/7514016/OpenHardwareMonitor.Report.txt)

namxam, 2021-11-23:
Yeah, I am seeing the same behaviour with the latest 0.9.7 release. Sensor 2 seems to be okay, the rest is showing values beyong 115˚C.

Is there anything we could provide to fix the issue? I have a few machines with the same config showing similar issues.

STARasGAMES, 2021-12-02:
Made me panic for a few minutes :)
[image]

Glad to know it's not dangerous. But would be cool if it would be "cool"... XD

handsomemans-rich, 2024-08-09:
[image]

This is Normal???

DarthDestroyer, 2024-08-09:
Check the Documentation of your MoBo where the Temp-Sensors are located!
HWMonitor and OHM are "only" reading what the Mainboard itself delivers - as you see, two Tools are measuring nearly the same Temp!

Maybe its located under the NVME or under the CPU or maybe at the Power-Parts - these are the Hotspots, i would say...
It depends! Look at the Docs or Manual where the Sensors are...

For sure, Temp Problems can lead to a BSOD - But that isnt the Fault from OHM