openhardwaremonitor/openhardwaremonitor #1098: Support for Asus ROG STRIX Z370-E Gaming

Kolthor, 2018-04-10:
Would love support for said motherboard, which has no sensors listed in v0.8.0.3-alpha:
[image]

BIOS lists the following temperature sensors:
- CPU Temperature
- Motherboard Temperature
- PCH Temperature
- T_Sensor Temperature
- EXT_Sensor1 Temperature
- EXT_Sensor2 Temperature
- EXT_Sensor3 Temperature

And the following fan speed sensors:
- CPU Fan Speed
- CPU Optional Fan Speed
- Chassis Fan 1 Speed
- M.2 Fan Speed
- Chassis Fan 2 Speed
- Extension Fan 1 Speed
- Extension Fan 2 Speed
- Extension Fan 3 Speed
- AIO Pump Speed

Attached report also generated using v0.8.0.3-alpha:
[OpenHardwareMonitor.Report.txt](https://github.com/openhardwaremonitor/openhardwaremonitor/files/1896061/OpenHardwareMonitor.Report.txt)

DarthDestroyer, 2018-04-10:
Chip ID: Unknown Winbond / Nuvoton / Fintek with ID 0xD121 at 0x2E/0x2F

I see your nice writing and i think, you will handle that ;)
#981 should help, tested in #1060

Kolthor, 2018-04-12:
PR #884 seems to do the trick, thanks!
[image]
(I renamed the fans manually, still need to figure out which temp is which)

Here's hoping it will make its way into the master.

sir-hashpipe, 2018-04-18:
Hi, I tried this as well for my Asus Z370 Prime-A and it works. However the VCORE seems to be divided by 2? I have same results as in picture above. It's always showing half compared to HWiNFO or HWMonitor. Anyone knows how to fix this? And is it possible to get each individual Core VID from CPU also like HWiNFO? Thanks

ThanOs47, 2018-09-24:
> PR #884 seems to do the trick, thanks!
> [image]
> (I renamed the fans manually, still need to figure out which temp is which)
> 
> Here's hoping it will make its way into the master.

Can you send me the file for download ?

moellerm, 2020-03-08:
This should be fixed in Open Hardware Monitor Version 0.9.2.

closed 2020-03-08 by moellerm