openhardwaremonitor/openhardwaremonitor #1100: No option for remote login  (version 0.8)

ebojali, 2018-04-13:
Hi, 
I don't hav any option for remote login, I tried all version on your webpage.

Thanks
[image]

DarthDestroyer, 2018-04-13:
What do you mean with "remote Login"?

Are you using a RDP Session?

ebojali, 2018-04-13:
yes I am but I am also connected via monitor locally. But I want to use the rig monitoring app on andriod as I've always done but I have to enable that option first before I can use the app on my phone.

ebojali, 2018-04-14:
Anyone supporting this? :)

DarthDestroyer, 2018-04-15:
if i would understand you, maybe...
i dont really know, what you want!

I give you some tips. 

1. if you are missing sensors or other Data, check if you had started as admin and after that, create a report and upload it in your issue!
2. the latest version is only available at github. 
3. rdp is and will ever be a problem! there are several issues! but i think, that it isnt relevant here. 

if you could upload a Report, that would help! 
It is somewhere at File>Report>Create/Save dont send it! Attach it here, please!

ebojali, 2018-04-15:
As mentioned in the installation instructions on this webpage:
http://rigmonitoring.net/home?locale=en

I did not get any RIG identifier and do not have the option to enable remote log.. I noticed that I wrote "remote login", sorry about that. I ment remote log of course.

If I downloaded the application linked from the webpage I got everything working, but not the version on github.

DarthDestroyer, 2018-04-15:
i think that you have to search in that application and not in OHM!

i dont know it, but you have to search there and not in ohm, maybe. 
that thing uses the library and does whatever it does with it! 

and if you download the zip from github, it is only the ohm Sourcecode! You have to compile it before you can use it... 

had you searched for help, at there forum / github or whatever they use?