openhardwaremonitor/openhardwaremonitor #1101: Support for ASRock J3455M

CHerSun, 2018-04-16:
Hi.

Could you add support for ASRock J3455M motherboard - a SoC based on Intel Celeron J3455? Running headless server on it and it'd be awesome to see temperature readings.
[OpenHardwareMonitor.Report.txt](https://github.com/openhardwaremonitor/openhardwaremonitor/files/1915131/OpenHardwareMonitor.Report.txt)

CHerSun, 2018-04-16:
Update: I've built latest version on github (0.8.0.5) - it sees processor readings correctly (was using 0.8.0 before). But motherboard shows no sensors at all still.

DarthDestroyer, 2018-04-16:
You have to add your MoBo Chipset from a Pull Request!

Chip ID: Unknown Winbond / Nuvoton / Fintek with ID 0xD121 at 0x2E/0x2F

If you search for the ID, you will find it! 
Im on mobile and at work... I think you will find it ;-)

CHerSun, 2018-04-22:
@DarthDestroyer I've tried to add the chip. While searching internet for the actual chip model I've found a pull request for this chip already made:
https://github.com/openhardwaremonitor/openhardwaremonitor/pull/884/commits/793888263494bb77be64aacff2250ed19dd0d545
But I cannot find this code in actual base. Was it not added?

CHerSun, 2018-04-22:
I've added changes to my copy from that pull request (using && instead of || in last if). MB chip was detected and read, but it looks like there are just too many sensors reported, not sure if that is from program or from chip reports.

DarthDestroyer, 2018-04-22:
There are a Lot of Sensors! As far as i had seen it looks good. (The Pull Request)

Well done, btw!

CHerSun, 2018-04-22:
@DarthDestroyer Tbh, I do not have experience with GitHub, but checking code at master branch I see that that pull request code is not present in master branch (I wouldn't have to search for that chip support otherwise). I've double-checked this. Also code at master branch for NCT and LPCIO modules are more than 2 years old without any recent changes. 

If you could merge all this code together - it would be great.

DarthDestroyer, 2018-04-22:
Okay, let me explain that!

The master includes all accepted and closed Pull Requests, all open Pull Requests are (like the name says) open and not yet accepted!

In the Pull Request itself is exactly said, what have to be changed that it works and atm, as @moellerm isnt available there is no other option as manualy adding that code to your own code! Sorry for that